prusa3d/PrusaSlicer #1795: New feature might really be a bug

jnadel, 2019-02-07:
### Version
1.41.2

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version windows
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
"In earlier versions, the support interface layers were printed as if all the object bottom layers were printed with a bridging flow. Starting with this version, an interface layer is generated to support the non-bridging overhang perimeters as well" THIS NEW FEATURE IS PROBLEMATIC IN CASES WHERE THERE AR LARGE UNSUPPORTED SPANS/CANTILEVERS. CAN YOU PLEASE MAKE THIS OPTIONAL? I tried to use the support enforcer object to overcome this but that seems to still generate support material in accordance with the new feature rules.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-02-07:
Please provide an image of where the support generator is problematic.

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv