prusa3d/PrusaSlicer #1798: First layer (including brim) floating in mid air

PaulKlinger, 2019-02-07:
### Version
Tested in 1.42.0-alpha5+win64 and 1.41.1+win64. In the earlier version the first layer is shown correctly just after slicing but jumps up from 0.2 to 0.5 when exporting gcode.

### Operating system type + version
Windows 10 - 1803

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 3.4.2-1360

### Behavior
After slicing first layer is at 0.5mm instead of 0.2mm. Speed for this layer is also set to the normal one instead of the reduced first layer speed. (I noticed this issue when the printer started rapidly moving to print the brim in mid air.)

[image]

#### Project File (.3MF) where problem occurs
[ball.3mf.zip](https://github.com/prusa3d/Slic3r/files/2842605/ball.3mf.zip)

FidelCapo, 2019-02-08:
I think this is because the object is so small that the slicer ignores the first layers. If I turn "Detect thin walls" (Print Settings - Layers and perimeters - Quality) on, everything seems to be all right. Can you confirm it? Thank you.

PaulKlinger, 2019-02-08:
Thanks, that seems to work.

It would be nice if this would be turned on automatically, or at least an error was shown if the first layer is not on the print surface. I can't think of any case where the current behavior wouldn't lead to a failed print.

bubnikv, 2019-03-01:
Yeah, it is a known issue we shall fix.

TorqueBias, 2019-06-28:
I'm so glad I'm not alone! I spent 3 hours trying to figure out what was wrong. Turns out that Cura has the same bug too.  After both slicers failed, I recreated the model twice, and recalibrated the Z axis before I figured it out.

What would the correct fix be? Even if the brim was laid down correctly, the first layer of the model still wouldn't print would it? Dropping the whole model down one layer would introduce dimensional problems. If that had happened I would have gone crazy trying to figure out why the finished print wasn't tall enough. Enable thin wall detection on the first layer only maybe? Even a warning that the first layer wasn't printable, and to check for thin walls would have helped a bunch. Thinking back, I had a thin edged vertical tube that wouldn't print once too. I'll bet this was the same issue.

rtyr, 2021-01-22:
User is now notified. Closing.

[image]

closed 2021-01-22 by rtyr