prusa3d/PrusaSlicer #1802: [1.42a5] Add modifier position origin centered on model center

CapnBry, 2019-02-08:
### Version
Version 1.42.0-alpha5+win64

### Operating system type + version
Windows 10 x64 Pro

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When a modifier lambda object is added to a model object (STL), the origin of the new lambda object is in the Z-center of the parent object. This makes positioning the lambda especially difficult as you need to know exact height of your object in order to manually calculate the position for the lambda.

For example. I have a model that is 32mm tall. I want to add a lambda box so the first 5.04mm has an extra perimeter so I right-click, add modifier, box. The 28.5mm cube looks good, and extends from Z=0 to 28.5mm. Attempting to shrink the box down to Z size of 5.04mm, the box now hovers in the air. To get the box to go from Z=0 to Z=5.04, It is 32mm/2 - 5.04mm/2 + Z = 0, solve for Z. If you don't know the Z-height of the parent model, it is especially frustrating to try to determine it experimentally and a nightmare for a platter full of objects. This is true of all the "Add..." child objects: Part, Modifier, Support Blocker, Support Enforcer.

**These child objects need to at the very least have their Z origin set to 0 when instantiated so the Z-position calculation needed is simply `child object height / -2`.** The way it works now is great because the relative positions of the parent and child are maintained throughout a translation and rotation of the parent object, but puts the onus of doing all the positioning math on the user. If the modifier box is instantiated with Z origin set to 0, then subsequent X Y rotations of the parent object may produce undesirable positioning of the modifier, but this can be mitigated by the user by simply adding the modifiers after X Y rotation of the parent.

I think ideally, the child objects anchor point should not be in the center of the child object, but rather at the lowest Z position on the object, so that a Z position of 0 would result in the object sitting on the bed. However, that would create an issue when rotating in X and Y as the object would rotate off-center but again this could be mitigated by adding modifiers after parent X Y rotation.

Sebastianv650, 2019-03-17:
To add an image which might be easier to understand than the text:
I need a modiefier for the vertical part sticking out of the rib. The rib itself is 3mm high, therefore I need a modifier which starts 3mm above the print bed. To get that, I need to do some math described by @CapnBry above and set Z to 1.5mm. Very unintuitive, even impossible if you don't know where to search for the solution.

[image]

Isometric view just for information:
[image]

CapnBry, 2019-05-14:
PrusaSlicer 2.0.0 RC still handles modifier meshes in the same way and therefore doesn't have a sane way of positioning them apart from the aforementioned math or trial-and-error imprecisely dragging them around. A redeeming quality of the new modifier mesh mechanism is that I know when I add a slab, it is generated with modifier height = 0.5 * model height. From that, set `origin Z` to `-orig_modifier_height+0.5*desired_modifier_height + desired_modifier_start`, and then set `size Z` to `desired_modifier_height`.

This can be simplified for the case where the user desires the modifier start at Z=0, set `origin Z` to `-orig_modifier_height` and `size Z` to `2*desired_modifier_height`, which will generate half the modifier under the bed, but is easier to calculate without launching additional software to calculate how to get it into position.

Actually, testing my placement math with a 24.65mm tall object and a 6mm slab starting at 3mm doesn't work. If I set `origin Z` to `-12.32 + 0.5*6 + 3=-6.32` and the `size Z` to `6`, the modifier doesn't take effect until *after* Z=3.12mm (first difference at Z=3.36mm) where it should take effect at Z=3mm. 🤷‍♂

bubnikv, 2019-05-15:
We know that the current behavior is not ideal, but we will not manage to
change it in the PrusaSlicer 2.0.0 final, sorry.

CapnBry, 2019-05-15:
Oh thank you, that's great to hear that this is known. Since 1.42 brought the modifier meshes front and center in the UI, I've been using them like crazy to create better prints with less print time penalties than changing settings for the whole build plate. Usefulness is definitely moving in the right direction and the efforts of you and the rest of the contributors are greatly appreciated.

Re-reading my comment from yesterday, I apologize that the tone is a bit rude. I must have come home from work a little grumpy.

ctilley83, 2019-06-11:
@bubnikv Doesn't seem this made it in PrusaSlicer 2.0 final. Implementing the "Place on Face" button on modifiers would work fine. This is a vital behavior.

bubnikv, 2019-08-01:
> Implementing the "Place on Face" button on modifiers would work fine. 

That will be part of the upcoming PrusaSlicer 2.1.0-alpha1.

closed 2026-09-02 by CapnBry