prusa3d/PrusaSlicer #1803: Wrong width of perimeter for solid anchors

supermerill, 2019-02-08:
### Version
_Version of Slic3r Prusa Edition used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-18:
@supermerill Thanks, I have merged the 2nd part with 3f74ec491e53074e5dd01bd1686d1c1625e511c7
for my life I was not able to track the first change. Would you please point me to the spot in the current master? Thanks.

supermerill, 2019-04-18:
Oh, sorry, It's only on my code, I made an error when i checked it. It was about a little thing i made to prevent the "margin" to go over the void when there are no infill

replacing 
`const double margin = scale_(EXTERNAL_INFILL_MARGIN);`
by 
```
double margin = scale_(EXTERNAL_INFILL_MARGIN);
if (this->region()->config().fill_density.value <= 0.) {
    coord_t max_margin = 0;
    if ((this->region()->config().perimeters > 0)) {
        max_margin = this->flow(frExternalPerimeter).scaled_width() + this->flow(frPerimeter).scaled_spacing() * (this->region()->config().perimeters.value - 1);
    }
    margin = std::min(margin, max_margin);
}
```

closed 2026-09-29 by prusaslicer-bot[bot]