prusa3d/PrusaSlicer #1804: Layer height modifier not working

MartinMue, 2019-02-08:
### Version
1.42.0 alpha5

### Operating system type + version
Windows 10 x64 

### 3D printer brand / version + firmware version (if known)
MK2

### Behavior
lambda box - Layers and Perimeter:
The Layer height setting does not change the actual layer height. Different perimeters are working correctly.

_Is this a new feature request?_
No
[tester.zip](https://github.com/prusa3d/Slic3r/files/2846633/tester.zip)

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

thni1703, 2019-02-14:
I have the same problem w10 64bit with MK3 with and without MMU2

bubnikv, 2019-03-01:
Layer height cannot be easily modified by a modifier mesh. It is actually a bug that the user interface offers this parameter for the modifier mesh.

Namely, there is no guarantee that the modifier mesh overlaps the whole layer, and Slic3r requires the whole object layer to be sliced with the same layer height.

thni1703, 2019-03-02:
so how do you go about setting the layer size change ?

grarup, 2019-07-09:
@bubnikv Why not simply use the height of the modifier mesh to calculate where the layer height changes needs to be made. Or make a special modifier that is two planes parallel with the print bed.

bubnikv, 2019-08-01:
>  Or make a special modifier that is two planes parallel with the print bed.

This is what we did in the upcoming PrusaSlicer 2.1.0-alpha1. Closing.

closed 2019-08-01 by bubnikv