prusa3d/PrusaSlicer #1805: --export-svg and --export-png are missing

VolumetricBio, 2019-02-09:
### Version
Slic3r 1.42.0-alpha5 - Prusa Edition

### Operating system type + version
Ubuntu 18.10

### Behavior
--export-png is listed as a feature in the README, but Slic3r describes it as "Warning: unknown option --export-png"

--export-svg used to be a working feature in the original Slic3r. Was this feature nixed in Slic3r Prusa Edition? If so, why?

I would like to run:
$ ./slic3r makerook.stl --export-png --layer-height 0.1 --output pngs-export.zip
$ ./slic3r makerook.stl --export-svg --layer-height 0.1 --output multilayer-svg-export.svg

_Is this a new feature request?_
No, this is a "where did these old features go" request?

bubnikv, 2019-03-01:
why do you need to slice into svg or png?
We implemented slicing into PNG as part of the SLA support.

VolumetricBio, 2019-03-01:
Thank you. I mentioned —export-png doesn’t work for me. I found that option in the man page but it doesn’t work. Could you please explain how to get it to work? I’m happy to update documentation.

SVG export is another good way Slic3r used to output vector graphics for later conversion or complex image transforms per layer. I am unclear why svg export option was deleted. I am hoping it could be turned back on relatively easily...?

bubnikv, 2019-03-04:
We removed that functionality during rewrite of that particular code from
Perl to C++.

Any code removed is good if it is not used by anybody. It is even good to
remove the code if it is used by a very tiny fraction of the users.

To justify the effort to re-implement the export to png and export to SVG,
I need to find out the use case for it. Therefore I am asking: What were
you using it for? Are you using it for FDM or SLA? Are you aware, that for
SLA there is now a set of zipped PNGs exported by default?

VolumetricBio, 2019-03-04:
Thank you. We are using Slic3r for SLA.

No I didn’t know pngs were created by default, the documentation is incorrect then. Please clarify what command to run in command line.

 Can you specify specific image resolution for rendering the PNGs? This is why we need SVG export which can be rendered to any size png accurately, so that we can render custom sizes of PNG to very high resolution (like 4K and 8K) without loss of quality.

If we cannot export SVG very easily, how can the user instead enter custom sizes for PNG export?

And can we get more detailed documentation to explain new default command line behaviors?

bubnikv, 2019-03-04:
>  And can we get more detailed documentation to explain new default
command line behaviors?

The command line interface is not finalized yet. It should behave very much
like the 1.41.3 or master.

> Can you specify specific image resolution for rendering the PNGs? This is
why we need SVG export which can be rendered to any size png accurately, so
that we can render custom sizes of PNG to very high resolution (like 4K and
8K) without loss of quality.

If sliced through the user interface, just install the "SL1" printer with
the install wizard and adjust the printer settings.

VolumetricBio, 2019-03-04:
I understand we could try to use the GUI, but we would like to move our SLA printer to use your software instead of the vanilla slic3r. Since we are developing our own open source touch interface we need the command line version. So we would need the command line version to have these features.

Could that be possible? We could help with Dev costs too. We really appreciate and support open source.

bubnikv, 2019-03-14:
This will be fixed in 1.42.0 beta

The 1.42.0-beta command line processing is based on the upstream code by @alexrj and @lordofhyphens, thanks to them.
The command line format is described by https://manual.slic3r.org/advanced/command-line
with some exceptions:
- instead of --help-options we have --help-fff and --help-sla
- we have an additional parameter --loglevel
- we support --export-sla, not --export-sla-svg or --export-svg
- we don't support --cut-grid --cut-x --cut-y --autosave

The output of the --export-sla is a zipped set of pngs. You can set the display resolution with 
display_width / display_height / display_pixels_x / display_pixels_y

closed 2019-03-14 by bubnikv

jonpry, 2020-01-02:
I also need SVG output. The use case is high resolution SLA. Work area in pixels is 40,000x60,000. It's tricky to do sizes like this without huge memory requirements, Prusa-slicer uses more memory than I have to attempt it. Currently we use a custom rasterizer based on cairo that implements some needed non-linear transforms. This is all pretty fast and the SVG is not overly huge.

In the meantime I have been trying to add supports in prusa-slicer and then export to svg in Slic3r. Unfortunately I am not able to get the exported STL w/ supports to successfully slice in Slic3r most of the time. Presumably Slic3r is having trouble with layers that have lots of support<->model joints. 

Imho for really large files BREP is the way to go. Voxels are just unmanageable.

VolumetricBio, 2020-01-02:
> I also need SVG output. The use case is high resolution SLA. Work area in pixels is 40,000x60,000. It's tricky to do sizes like this without huge memory requirements, Prusa-slicer uses more memory [...]

Wow that’s awesome. Could you explain more on what you’re doing at so many pixels (40,000x60,000)? What material are you using, what optical system, and what 3D printer? Is it all custom? Let’s connect to discuss more for sure.

pinojojo, 2022-01-27:
My situation is similar, I use a high magnification optical system, each layer may contain 50k*50k Pixels, each layer must be divided into many small pieces to print in chunks. So it would be nice to have an svg file that allows us to customize the rasterization parameters

bubnikv, 2022-01-28:
@tamasmeszaros works on extending the SL1 format, where instead of PNGs, SVG files are generated. The exported files are somehow smaller than PNG containing archives, however don't expect miracles. I think @tamasmeszaros reported around 5x smaller files, but the reduction ration depends on allowed contour decimation rate.

tamasmeszaros, 2022-02-07:
Exporting SLA output into an archive with SVG images inside: https://github.com/prusa3d/PrusaSlicer/commit/3785ad3de494dae7888c8d97a0ee5aa38861e69e

Please note that this is currently an experimental feature and the controlling paramters (`sla_archive_format` and `sla_output_precision`) may change.

Currently, in order to actually export the svg-s you need to set `sla_archive_format` to value "SL2". 

In order to use this from the command line, you need something similar to:
```shell
prusa-slicer --display-pixels-x X --display-pixels-y Y --display-width W --display-height H --export-sla /home/quarky/Downloads/Marvin_KeyChain.STL --sla-archive-format SL2 --output ./Marvin_KeyChain.zip
```
Given that you substitute your values for X, Y, W and H. Also to achieve correct results, the print bed shape end center point should be set properly. All of these parameters can be part of a custom printer profile you create and use subsequently. (see the `---load` command line option)

pinojojo, 2022-03-28:
@tamasmeszaros  hi, I used the 2.5.0 alpha to test a model which has few holes in it, but I found that the exported svg is not composed of polygons like the old slic3r lib which mark each polygon's type (hole/contour) . would you consider mark the type of each path(or polygon) when exporting svgs?

pinojojo, 2022-03-29:
@tamasmeszaros  hi, could you make exported svg format like the old slic3r export svg feature which each polygon has a type attribute that declare the polygon is a contour or a hole.  I tested the SL2 output svg option and found that when slicing model has holes in it,  types of the nested polygons is not marked explicitly.

testing with a model below, and use `prusa-slicer --display-pixels-x 1000 --display-pixels-y 1000 --display-width 200 --display-height 200 --export-sla UMS5_Fresnel_lens.stl --sla-archive-format SL2 --output ooo.zip
[UMS5_Fresnel_lens.zip](https://github.com/prusa3d/PrusaSlicer/files/8367564/UMS5_Fresnel_lens.zip)
`

pinojojo, 2022-03-30:
Refer to this document [https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/fill-rule](url) when exporting nested paths. `fill-rule` should be defined. or by default` fill-rule` which means ` non-zero` , path direction should be correct.