prusa3d/PrusaSlicer #1806: Feature request: Ability to disable gap fill if segment is under a certain user-configured length

glowtape, 2019-02-09:
### Version
1.42.0-alpha5+win64

### Operating system type + version
Windows 10 Pro 17763.292

### 3D printer brand / version + firmware version (if known)
Voron 2.1 + Klipper 0.7.0

### Behavior
Short segment gap fills may cause artifacts depending on geometry and printer setup. I've a long print going and I'm observing this effect currently. The issue is probably more prevalent with a bowden setup, like mine. The layer is being laid down, and near the end of the layer, up to a certain height, tiny gap fills are being done. The printer however tends to over-extrude these (the segment probably doesn't gel with linear/pressure advance or vice versa), creating blobs slightly higher than the layer height, into which the nozzle eventually bumps into on the next layer. Since they cooled down too much by then, it moves the nozzle instead of getting ironed it away.

See this example from the slicer:
[image]

It would be interesting to have an option to drop gap fills, if the segment is under a certain length.

I will be attaching a picture of the finished print once it's done.

_Is this a new feature request?_
Yes.

#### Project File (.3MF) where problem occurs
[Vent Thingy 3.zip](https://github.com/prusa3d/Slic3r/files/2847867/Vent.Thingy.3.zip)

thiagopeixoto16, 2020-05-08:
👀 Looking for this option...  at least the Slic3r option to **disable Gap fill**

neophyl, 2020-05-08:
To disable gapfill you currently have to set the gapfill speed to 0.

thiagopeixoto16, 2020-05-19:
omg, thank you very much @neophyl ! I didn't realize that!

closed 2026-09-29 by prusaslicer-bot[bot]