prusa3d/PrusaSlicer #1807: Toolchange Custom GCODE Not Exporting

BryanSmithDev, 2019-02-09:
### Version
1.42.0-alpha5

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3 running FW 3.5.2-RC1 and MMUv2 on 1.0.4-RC1

### Behavior
Slicing with a MMUv2 Profile (in this case the stock one included in the Slic3r config), any commands entered in the Toolchange Custom GCODE on the Printer Settings Tab never make it to the exported GCODE.

Expected Results: 
If M300 is entered in the Toolchange GCODE box, then the exported GCODE should have M300 at every toolchange.

Actual Results:
 M300 only occurs once in the exported GCODE at the bottom of the file where all Slic3r Params are exported:   ; toolchange_gcode = M300

Is this a new feature request? No

#### Project File (.3MF) where problem occurs
This occurs regardless of models used and I can also replicate it on older versions of Slic3r PE. Including prior to UI update.

Is this intentional for MMU? I don't see why it would be. And if it is, the toolchange GCODE box should be disabled similar to how it is on single extruder profile.

joonadec, 2019-02-20:
Hello, I ran into this problem myself...I believe it doesn't apply the contents of the 'Toolchange GCODE' because the tool doesn't change i.e. it is using the same nozzle.

Currently am using a work around, by applying the Gcode I need (am using a Y-splitter to multiplex filament) into the 'Before Layer Change' box instead.

lukasmatena, 2019-06-27:
Yes, we are currently working on an improvement. If it passes all testing, it will be in master branch soon.

rtyr, 2020-03-29:
Already implemented (since PS 2.1.0-alpha1). Closing.

closed 2020-03-29 by rtyr