prusa3d/PrusaSlicer #1808: adaptative layer not working

Tinchus2009, 2019-02-09:
### Version
latest alpha 5

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
custom prusa i3 + marlin 1.1.9
### Behavior
Im trying to choose the adaptative layers feature. Max range it is 0.3 minimun range is 0.1
I do the editing, but when I slice, I only get 0.3 layers all across the model
[EndCapShoulder.zip](https://github.com/prusa3d/Slic3r/files/2848277/EndCapShoulder.zip)

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-03-01:
This is interesting. When I loaded your 3MF file, no variable layer heights were shown in the sliced G-code, even though the variable layer height editor showed that you modified the profile. Even if I modified the profile, the G-code still shows no variable layer height.

It started to work after I reset the profile in the profile editor and I painted a new profile.

I have to dig into your 3MF to find out what went wrong.

Tinchus2009, 2019-03-04:
Yes, exactly that is what I saw. Separate question: how can I get in contact with you regarding a a development proposal?

bubnikv, 2019-03-04:
>  how can I get in contact with you regarding a a development proposal?

What do you mean with the development proposal?

Tinchus2009, 2019-03-05:
I work for a company. We would like to make some development based on
slic3r. We need a developer to do that. I was thinking on you.

bubnikv, 2019-03-06:
I am with Prusa Research. I am afraid we have enough of our work to do. If
it is useful to the general audience, we may implement it anyway.

bubnikv, 2019-04-13:
I have fixed #2073
"Variable layer height feature breaks after rotating part"

Would you please retest with the upcoming beta2, whether it helps with your issue? Thanks.

bubnikv, 2019-05-04:
no response, it should be fixed. closing.

closed 2019-05-04 by bubnikv