prusa3d/PrusaSlicer #1809: Wrong perspective in the slicer preview

lunicorn-x64, 2019-02-10:
### Version
1.41.2+

### Operating system type + version
Mac OS and Windows

### 3D printer brand / version + firmware version (if known)
unneeded information

### Behavior
The 3D view of the slicer is wrong in perspective. 
It should be wider at the front than at the back. 
Currently it is wider in the back than in the front.

#### Wrong
[image]

#### Right
[image]

bubnikv, 2019-02-10:
What you see is a correct orthographic projection. The rear edge is as long
as the front one. It is only an optical illusion, that the rear edge seems
to be longer than the front one.

lunicorn-x64, 2019-02-10:
Yes, but that's exactly the fault that it's the same length in the back and in the front, Because in a 3 dimensional environment you need a perspective.

Sebastianv650, 2019-02-10:
You absolutely don't need a perspective as you have in mind. In fact, if you use 3D views for engineering (CAD and similar) you usually work in orthographioc view. In most cases you can switch between the two modes, but as "your" mode is disorting dimensions it's not good for use except renderings.
For exactly this reason, Slic3r should stick to the way it handles 3D views.

lunicorn-x64, 2019-02-11:
okay. So wo it be possible to add a perspective mode? so you could switch between?

closed 2019-02-11 by vojtechkral