prusa3d/PrusaSlicer #1810: Crash when trying to split (repaired) model

torandi, 2019-02-10:
### Version
1.41.2+win64 Prusa Edition

### Operating system type + version
Windows 10 Pro 1803

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3

### Behavior
After importing this model, and getting "200 errors repaired", I try to use the "Split" button. After a short hang the entire application crashes.

[Marid.stl.zip](https://github.com/prusa3d/Slic3r/files/2849013/Marid.stl.zip)

BerndJM, 2019-02-10:
There are apperently more errors in this model than the automatic repair can fix.
The attached file is run through an external repair-service and splits fine.
You will recognice that there are some tiny artefacts - not visible in the 3D view ...

[Marid (repaired).stl.zip](https://github.com/prusa3d/Slic3r/files/2849522/Marid.repaired.stl.zip)

bubnikv, 2019-03-01:
The upcoming Slic3r 1.42.0-alpha7 integrates a robust hole triangulation algorithm. While it may not produce a valid triangulation for a broken model, it should not crash Slic3r.

Closing.

closed 2019-03-01 by bubnikv