prusa3d/PrusaSlicer #1812: Nozzle with offset. Incorrect preview after generating g-code

barskykd, 2019-02-11:
### Version 
1.42.0-alpha5+win64

### Operating system type + version
Windows 10

### Behavior

I have dual nozzle (e3d chimera) printer. 
For the second nozzle I set offset in settings.

[image]

After slicing preview shows toolpath for second nozzle without offset. G-code itself is ok.

* _Expected Results_
[image]

* _Actual Results_
[image]

bubnikv, 2019-03-01:
@enricoturri1966 this is something for you. Would you please add the extruder offset to the G-code preview? Thanks.

enricoturri1966, 2019-03-05:
Working on it -> http://dev.prusa3d.com/browse/SPE-842

FidelCapo, 2019-03-08:
This issue should be fixed in next release. Thank you for report.
Closing

closed 2019-03-08 by FidelCapo