prusa3d/PrusaSlicer #1818: Change of object type support blocker to enforcer does nothing (and vice versa)

ochm, 2019-02-13:
### Version
1.42.0-alpha5 w64

### Operating system type + version
W8.1

### 3D printer brand / version + firmware version (if known)
MK3

### Behavior
When added Support enforcement object and later changed type to Support blocker,
the object remains still as Enforcement (and vice versa).
The only way how to change object type is create new one and delete old one.

#### Project File (.3MF) where problem occurs
[test.zip](https://github.com/prusa3d/Slic3r/files/2861396/test.zip)

FidelCapo, 2019-02-19:
Unfortunately, I am unable to reproduce this issue. If you can attach short video, it could help.

ochm, 2019-02-20:
> does nothing (and vice versa) (#1818)
> https://github.com/prusa3d/Slic3r/issues/1818#issuecomment-465029900 | view it
> on GitHub ] , or [
> | mute the thread ] .

Hi, 

video is here https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAa9rC6cfoScaz-o4tS5I_PT1TRGiATCtKaoQV6jNzoyS5gqZdZtsaINc%2F%24%7Bf%7D%3Fo%3DAlFD4P8KsnX7qgfXK20VRw_W4Gg4XneEByhB5MA1SSyh%26v%3D1%26x%3D3%26a%3DBwOUgSrTwNc6rpat1q7L7Rup3UyzA5sASgEACAHIAP9MUomRA9tXSQ%26e%3D1553293125%26k%3D%24%7Buk%7D%26fl%3D%26r%3D252014D5-A6C0-4A71-B1B8-F41B60BEBAED-1%26ckc%3Dcom.apple.largeattachment%26ckz%3DB60CC2FD-66EB-41B1-BBE7-E000CAD0E12C%26p%3D25%26s%3DIGPre1T8khhdWrTWZgLvBJSTaUE&uk=Yv-mgkeZeHhIMktmabEHVg&f=IMG_4188.MOV&sz=45792337

At first is placed object with type support enforcement, after reslicing the supports are visible. 

Next is changed object type to support blocker, but after reslicing the supports are still in place. 

Is it clear now? 

Sincerely

FidelCapo, 2019-02-21:
Yes, now I can reproduce this issue. Thank you for report.

Created JIRA 829

bubnikv, 2019-02-21:
Thanks for reporting, fixed with c86a4f3ac6a5882b4da24fa0ee3b3e79a9af5053
Closing.

closed 2019-02-21 by bubnikv