prusa3d/PrusaSlicer #1819: Top and bottom infill are out of phase

aemileski, 2019-02-13:
### Version
1.41.2+win64

### Operating system type + version
Windows 8.1

### 3D printer brand / version + firmware version (if known)
Stock i3 mk3 + firmware 3.5.1

### Behavior
The top and bottom solid infill angles are 90 degrees out of phase, so the top and bottom don't match.

This wasn't always the case.  I don't remember the version when it changed, but I was sure it was obvious enough that someone else would have reported it already.  I guess I was wrong.

Steps to repeat:

1.  Look at the bottom infill pattern angle.
2.  Look at the top infill pattern angle.
3.  Notice that the two patterns are out of phase by 90 degrees.
4.  Change the number of top and / or bottom layers.
5.  Notice that the top and bottom patterns remain 90 degrees out of phase.
6.  Change the infill angle.
7.  Notice that the top and bottom patterns remain 90 degrees out of phase.

supermerill, 2019-02-14:
can't reproduce.

intended behavior:  
if layer_id %2 == 0 then angle = angle_setting
if layer_id %2 == 1 then angle = angle_setting + 90

aemileski, 2019-02-14:
Wild guess:  layer_id is indexed from 0.

2 Layers:  Bottom = 0, Top = 1

200 layers:  Bottom = 0, Top = 199

supermerill, 2019-02-14:
yes ><

bubnikv, 2019-03-01:
@aemileski Is it clear now? Would you please close this issue? Thanks.

aemileski, 2019-03-01:
I still disagree that is isn't a bug.

The angle should be relative to the layers-from-surface, not the absolute layer count.

closed 2019-03-01 by aemileski