prusa3d/PrusaSlicer #1821: Error on start

weirdoku, 2019-02-13:
"Preset bundle `PrusaResearch` version not found in index: 0.4.0-alpha3"

Sorry if this is not how to submit an issue but I am totally at a lost why it's not working. It was working last night but not today. The above line is the error that pops up when I open this version. The latest 1.41.3 version opens fine.

Zemistr, 2019-02-14:
The same 😭

vojtechkral, 2019-02-14:
From #1760:

To fix it, please remove either the whole configuration directory or the `cache` subdirectory inside the config directory.

The configuration directory is
On Windows in `%APPDATA%\Slic3rPE-alpha`
On Linux in `~/.Slic3rPE-alpha`
On Mac in `~/Library/Application Support/Slic3rPE-alpha`

weirdoku, 2019-02-14:
> From #1760:
> 
> To fix it, please remove either the whole configuration directory or the `cache` subdirectory inside the config directory.
> 
> The configuration directory is
> On Windows in `%APPDATA%\Slic3rPE-alpha`
> On Linux in `~/.Slic3rPE-alpha`
> On Mac in `~/Library/Application Support/Slic3rPE-alpha`

I LOVE YOU

vojtechkral, 2019-02-14:
Well, thanks, but this wouldn't happen if I hadn't screwed up in the first place :)

I'll leave this issue open to increase visibility and as a reminder to improve the update check procedure a bit so that this wouldn't happen in the future...

closed 2019-04-13 by vojtechkral