prusa3d/PrusaSlicer #1823: Alternating perimeter layers (hexagonal walls)

espr14, 2019-02-14:
Left side of image is standard layering (cut in x-z plane). Right one is newly proposed one. Outside wall/shell/perimeter (first column from the right) keeps the same layers but even columns are shifted by half of layer in z dimension. Each string is touching others in 6 spots (hexagon) instead of 4 in the standard one. This will provide higher strength of the printed piece. Can you add this to the slic3r? Thanks.
[image]

mdealer, 2019-03-30:
I would suggest to to move the alternating perimeter inwards by one to improve external appearance.

espr14, 2019-03-30:
Only even columns will be modified.

Mirarkitty, 2021-01-13:
I had basically the same idea  (#5767, doh). This should work very well for z-axis adhesion.

mvadu, 2021-01-13:
to be frank in a well leveled, well working printer, the extruded filament should get little squished (so cross section will be more elliptical than circular) thus touching more than 4 small points like your left image. But I agree having an interleaving on z should at least in theory have a better layer adhesion. More of CNCKitchen's Stephan to test it out once Prusa slicer implements it.

Mirarkitty, 2021-08-30:
Just had a layer separate, looking forward to this...

Plutoaurus, 2024-02-17:
Here after watching cnc kitchen. 
Thus looks to improve layer adhesion quite a bit.
+1 request

dicksondickson, 2024-02-17:
Also here after watching CNC Kitchen.

Ascendzor, 2024-02-17:
I have tried to research this to make my prints more watertight.
+1 this request. Would help me greatly

Comprehensive-Jason, 2024-02-18:
Here's the relevant CNC Kitchen video: https://youtu.be/5hGm6cubFVs

This technique seems to improve print strength quite significantly with no serious downsides, so would be a great addition!

SaltyPaws, 2024-02-18:
+1. To this request after seeing Stefan's test results!

Mirarkitty, 2024-02-18:
Nice Stefan. Hopefully this will get some traction now!

migueldiaslima, 2024-02-18:
Hoping this gets implemented on prusaslicer!

Comprehensive-Jason, 2024-02-19:
Possible parasitic patent? The patent is currently assigned to Addman Holdings, which advertises it as their technology to increase 3D print strength: https://www.addmangroup.com/wp-content/uploads/2023/09/ADDMAN_CAAM_Flyer-8.5x11.pdf

alxpettit, 2024-02-20:
If the patent was submitted after, we should be good to implement this, yes?

Comprehensive-Jason, 2024-02-20:
> If the patent was submitted after, we should be good to implement this, yes?

It depends on whether this issue, or earlier proposals like it, [can count as "prior art"](https://www.uspto.gov/sites/default/files/documents/Basics-of-Prior-Art-Searching.pdf) made before this patent.
Either way, I still think implementing is worth a shot. Patenting something this simple is retarded and a scourge on human progress, and the feature can always be "removed" if Addman Holdings gets ban-happy (someone else will just fork it once the cat's out of the bag).

kevinmorrison, 2024-02-21:
Would love to see this implemented!

https://youtu.be/5hGm6cubFVs?si=SALNuTnXxMOwcJ7f

astraiophos, 2024-02-21:
+1 for adding this feature to the slicer.

The patent issue is an important concern, but the feature request was made before the patent filing, so the idea existed before the patent. They'd have a hard time justifying the ban based on the patent alone since there's history of wanting this feature before the patent was filed. I'm no lawyer though, so that's just my opinion.

Harvie, 2024-02-23:
turn the pattern by 90 degrees (half width on sides, instead of half height at bottom and top). i think it might seem counterintuitive, but stronger. because it would allow plastic to be squeezed harder in those cracks...

Comprehensive-Jason, 2024-02-23:
@Harvie This is definitely another approach that could be fruitful and easier to implement. I am concerned this will have a more significant effect on part appearance though, as we run the risk of introducing vertical striping.

Why not both, I guess?

Harvie, 2024-02-23:
> I am concerned this will have a more significant effect on part appearance though, as we run the risk of introducing vertical striping.

Actualy i think this is concern for both approaches. And for both it can probably be fixed by lowering layer height.
Also that's why we have slicer options. To give user choice between looks and durability. I mostly do practical/functional prints. Some people do purely visual trinkets and action figures. Therefore we need to tune our slicer settings differently for these use cases.

dewi-ny-je, 2024-03-07:
> @Harvie This is definitely another approach that could be fruitful and easier to implement. I am concerned this will have a more significant effect on part appearance though, as we run the risk of introducing vertical striping.
> 
> Why not both, I guess? The vertical stagger approach might result in better vertical strength, while the horizontal stagger approach may produce better horizontal strength.

In my opinion the idea of staggering horizontally was wrongly drawn: you should not use thinner extrusion in the odd layers of the proposed image, but rather 1 1/2 wide extrusions. It will also be easier to avoid striping, since controlling 50% larger extrusions is more accurate than trying to extrude half as much.

Harvie, 2024-03-08:
> you should not use thinner extrusion in the odd layers of the proposed image, but rather 1 1/2 wide extrusions

That is quite genious! As long as the nozzle allows for that. I have some chinese 0.6mm nozzle, which does not have very wide face. It really sucks for ironing, so i should probably upgrade anyway. Might be an issue tho. Also i think your idea might be applied to layer height as well (as long as it's not already maxed out).

dewi-ny-je, 2024-03-08:
> That is quite genious! As long as the nozzle allows for that. I have some chinese 0.6mm nozzle, which does not have very wide face. 

If your nozzle doesn't allow it, then use normal stacking. It's not a mandatory feature, whenever it will be available.

Using half-wide extrusion is NOT the way to do it to get any reasonably good result (extruding 0.25 mm with a 0.4 mm nozzle??) and it should not even be offered as option.

The software can take over weaknesses of the hardware, but when that's the only way.
Changing nozzle is so simple that the software should not be made more complex just to save 2 euro/dollars, and for an optional feature anyway.

It's a matter of principle, future maintenance costs for the slicer developers, ...

Harvie, 2024-03-08:
Anyway. I've updated the mockup as per your suggestion:

[image]

DarwinAnim8or, 2024-03-27:
> > If the patent was submitted after, we should be good to implement this, yes?
> 
> @alxpettit It depends on whether this issue, or earlier proposals like it, [can count as "prior art"](https://www.uspto.gov/sites/default/files/documents/Basics-of-Prior-Art-Searching.pdf) made before this patent. Either way, I still think implementing is worth a shot. Patenting something this simple is retarded and a scourge on human progress, and the feature can always be "removed" if Addman Holdings gets ban-happy (someone else will just fork it once the cat's out of the bag).

Fully agreed with you here-- I don't think a patent for what is effectively a primitive building technique we've been using for thousands of years should ever be considered valid.

dewi-ny-je, 2024-03-27:
> Fully agreed with you here-- I don't think a patent for what is effectively a primitive building technique we've been using for thousands of years should ever be considered valid.

That's not how patent work. Applying something known even for a ping time from a field (building construction) to a different and not similar field (plastic and 3D fabrication) may very well be considered as an inventive contribution and therefore deserve a patent.
Of course if the patent was broad and is valid also for concrete 3D printing, then the field of application becomes close enough that the technique known for centuries can be applied without needing the innovative combination.

Louis2747, 2024-04-20:
> > Fully agreed with you here-- I don't think a patent for what is effectively a primitive building technique we've been using for thousands of years should ever be considered valid.
> 
> That's not how patent work.

i don't think they were talking about how patents work, i think they were talking about how they _should_ work.
The former is what lawyers are for, the second is a matter of opinion.

AbitNutzT00, 2024-04-27:
I don't understand why this isn't being implemented. To those of us who print 3D airplanes every bit of increased strength can be used to lighten the airplane. This makes this more important than just a minor increase in strength.

Sakura-sx, 2024-04-27:
> I don't understand why this isn't being implemented. To those of us who print 3D airplanes every bit of increased strength can be used to lighten the airplane. This makes this more important than just a minor increase in strength.

@AbitNutzT00 Then check SuperSlicer, they are usually the first at implementing these kinds of things.

dahappychappy, 2024-05-12:
I would love to see this implemented please. Great work everyone involved. Makes total sense.

weekendkoder, 2024-07-08:
This is why patent-scouts exist :-) . Anyways. Somehow its like someone make a patent on "how you stack" building-bricks together. 

Workaround: If you have specific tools inside a slicer (lets say an "offset feature"), the user can use it like he wants. F.e. shift every n-layers with 0.5 to  direction x-y-z and so on.  This way it does not collide with the patent. It means the slicer-software hands out additional abilities to users that have new ideas for new methods. The user could implement such offset functions / or macros by himself, which goes to a similar direction of "alternating layers".

Just my thoughts ;-)

dewi-ny-je, 2024-07-08:
> Workaround: If you have specific tools inside a slicer (lets say an "offset feature"), the user can use it like he wants. F.e. shift every n-layers with 0.5 to direction x-y-z and so on. This way it [...]

I'm not sure it works this way, since the "method" of stacking would be protected as well and if you have such a setting you are performing the patented method.

But it doesn't matter, as I mentioned the patent exists only in US and it's likely not even valid.
The software (assuming the main development takes place outside Europe) could be distributed anyway and the feature (if there is a complaint) could be disabled in US.
Like, the user has to pick their country upon installing it (wink wink...).

But anyway, the patent is generic enough to cover concrete 3D printing and stacking bricks with an offset is known for ages, so the patent is likely invalid.
Assuming they even try to enforce it.

weekendkoder, 2024-07-08:
What i meant with my comment: prusaslicer dont need to implement anything what goes into this direction. They just give the users more control about gcode generation. So there is no 1x click button / solution. Nothing. But the user will be able to modify his gcode like he wants. As he is already able to do it via notepad f.e. But its too much work to do it by hand. This is what i meant by adding more features to macros / gcode algorithm.

Anyways ... like you said, maybe its a different situation if the company is based in EU.

doudar, 2024-10-21:
I think another way that this could be implemented is by implementing vertical fuzzy skin on horizontal surfaces, like between layers. It might be slow, but should result in a stronger bond between layers. Depending on the amount of "fuzziness", you might even be able to ignore the previous layer when laying down the next, relying on flow to fill the valleys and peaks.

Harvie, 2024-10-21:
Actualy i was thinking that current implementation of fuzzy skin creates those tiny gaps between outermost perimeter and second outermost... If all of them were fuzzy, it might be stronger. (inner perimeters can use some kind of low-pass filtering to only reflect the big ripples and ignore the sharp corners. that way we can remove 90% of gaps, especialy the big ones, without introducing unnecessary acceleration and vibration to the machine...)

rasikrodri, 2024-11-08:
About the patent.
It seems that Stratasis already jad a patent for this and it expired on 2015.

If this is the case the new patent should be invalid.

All the info it's here https://www.patreon.com/posts/115566868

kompumaster, 2024-11-08:
Video about patents.
https://www.youtube.com/watch?v=9IdNA_hWiyE

weekendkoder, 2024-11-09:
@kompumaster ...thank you, for the informative video-link.
@rasikrodri ... i have the same view on this. It expired 2015 "and it already was public". The new patent rely on the old one. So it cant be valid. I am not a lawyer. But from a logical perspective, something seems to be wrong with the new patent. (Otherwise the patent system needs to be overhauled)

crabdancing, 2024-11-10:
@weekendkoder I mean, the patent system already needs to be overhauled, but I get what you're saying. :P

kompumaster, 2024-11-10:
If someone can implement this feature, just post the code in Russian GitFlic, or Chinese Gitee/Gogs. And no problems with patents 😁

kompumaster, 2024-11-10:
..or name the option "bricklaying" and let them try to challenge the patent "invention" of this technology.
[image]

lugui1998, 2024-11-10:
> If someone can implement this feature, just post the code in Russian GitFlic, or Chinese Gitee/Gogs. And no problems with patents 😁

Im not a lawyer, but I think it is relevant to this thread the information pointed and very well explained by Geek Detour on this video: https://youtu.be/9IdNA_hWiyE?t=196
TL;DR; The [current existing patent](https://patentimages.storage.googleapis.com/6f/10/98/8e9ed53fc39b27/US11331848.pdf) is possibly (and very likely) invalid and should have never been validated by the patent office, as there is already a [preceding one from 1995](https://patentimages.storage.googleapis.com/ea/e7/3c/2b836c9a51e18b/US5653925.pdf) which **already expired**. That would mean this technique is already in public domain.

dewi-ny-je, 2024-11-11:
Unless someone with a legal background in IP wants to contribute, everyone else please stop spamming with personal opinions: this is a thread about a feature and no one is talking about implementing it.

prochac, 2024-11-11:
Anyone can create an GitHub account, and anyone can open a PR, without it being merged.
But if this issue will be mentioned, it will pop out here.

But be really sure it's legally ok :wink:

dewi-ny-je, 2024-11-11:
@Sakura-sx Prusa is based in Europe so they are not bound by US patents.
In Europe the content of the patent granted in US has been (correctly) recognised as already known so the inventor/applicant is trying to patent the idea of this thread.
Hopefully they'll realise this idea too was know before the guy tried to protect it

In general, Prusa is safe either way

dewi-ny-je, 2024-11-11:
This one is a good idea though and they'll likely get a patent everywhere 

https://patents.google.com/patent/US20240217167A1/en

prochac, 2024-11-12:
> This one is a good idea though and they'll likely get a patent everywhere 
> 
> https://patents.google.com/patent/US20240217167A1/en

Isn't it about the shape? If I choose a different angle/shape, can I patent it too?

probonopd, 2024-12-20:
https://www.youtube.com/watch?v=9IdNA_hWiyE challenges the more recent patents and mentions that this method was patented by Stratasys in 1995, and hence the method should be Public Domain now. The author concludes that "the 2020 patent must be challenged".

Unkleskunky, 2024-12-23:
If anyone is implementing this feature by way of settings I would also like to implement. We should feel free to do so given the information we have here and would be unified in our contempt of the vulture patenting.

mimi89999, 2024-12-24:
Hello,
I saw that there was an attempt to implement this in https://github.com/prusa3d/PrusaSlicer/pull/10429, but the PR was closed with the following comment (https://github.com/prusa3d/PrusaSlicer/pull/10429#issuecomment-1890374482)
> also now I don't think it indeed increases strength too much

I'm wondering whether some tests were made to compare the strength of this

Unkleskunky, 2024-12-25:
> closed with the following comment (#10429 (comment)

I saw figures with claims of 10-14%.

Ruakij, 2024-12-25:
@mimi89999 Though it looks like that PR is rather changing extrusion width instead of height as described in the Issue here, basically trying to get stronger layers by interlocking horizontally instead of vertically.

I would be interested in seeing a comparison of both variants and maybe even both combined.

dewi-ny-je, 2024-12-25:
That PR was patented in 1995 and helps to fill gaps between extrusion lines, it should be easy to implement and brings lower porosity.
It could be the start.

tilmaen, 2025-01-11:
I posted the following in the Cura GitHub already:

I would absolutely love this feature.
The European patent hasn’t been granted yet (it’s been in limbo for four years). It seems to be stuck due to a more thorough examination process. Third-party observations have been filed (and acknowledged), and the examiner appears to have identified several issues related to "lack of novelty." Additionally, the incorrect references to prior patents were criticized by the EPO officer (unlike the USPTO).

The correspondence is quite interesting: [https://register.epo.org/application?number=EP21175830&lng=en&tab=doclist](https://github.com/Ultimaker/Cura/issues/url).

I have a friend who works there (in a different department), and I might ask him if he can share any insights about this patent application.

probonopd, 2025-01-11:
So in the official patent filings (File: 21175830.5 Non-patent literature filed by a third party) there is an attachment showing the headline of this PR (from 2019). Interesting...

tilmaen, 2025-01-11:
> So in the official patent filings (File: 21175830.5 Non-patent literature filed by a third party) there is an attachment showing the headline of this PR (from 2019). Interesting...

exactly what i thought :D

V037, 2025-01-13:
Guys... Instead of trying layer adhesion strength from the vertical axis why don't you try the strength from other sides... Maybe that extra layer adhesion will drastically improve strength on the perpendicular side of the layers

V037, 2025-01-13:
I also want to see this function so badly... Please someone make this and get it work with orca slicer or prusa slicer

dewi-ny-je, 2025-01-13:
The horizontal shifting of the perimeters to fill the gaps between lines of the lower layer would be already very useful

JirikP, 2025-01-23:
It has been released as python script. It seems that the patent from patent troll is not valid in EU.
https://github.com/TengerTechnologies/Bricklayers

[11 more replies]