prusa3d/PrusaSlicer #1824: [Feature Request] Add printing temperature to modifier options (and maybe EM as well?)

th33xitus, 2019-02-14:
Simple feature request to add the printing temperature as changeable value to the modifier.
That would be the first step of implementing a function how to generate a temperature-tower directly in the software without the need of post processing the gcode either by hand or with python scripts or whatsoever.

Another thought is to add the extrusion multiplier as a changeable value as well.
There are probably people out there who would like to intentionally under/overextrude specific parts of their prints.

ps: when will we be able to uses processes like in S3D? :P :D jk

bubnikv, 2019-02-17:
> ps: when will we be able to uses processes like in S3D? :P :D jk

Please open another issue and describe in as much detail as possible what
you expect from the "processes"

CPHfx, 2020-11-17:
Hi, 
I would like to add to this feature request. I am also missing the option to change the extrusion multiplier within the object (by using modifiers).
I described the reason for trying this [here](https://forum.prusaprinters.org/forum/prusaslicer/extrusion-multiplier-modifier-2/).
I think, this is addressed by this request - or do you want me to open a new one with a single one (only extrusion multiplier?)
All the best

rtyr, 2020-11-24:
Duplicate of your https://github.com/prusa3d/PrusaSlicer/issues/3635. Closing.

closed 2020-11-24 by rtyr