prusa3d/PrusaSlicer #1826: [Feature Request] SLA Mode Supports: Minimum X/Y Distance

Maeyanie, 2019-02-15:
I realize SLA mode is still very much WIP, and this may already be on the roadmap; I did notice there is a "lm_sla_supports_auto2" branch already, but that doesn't seem to cover this at the time of writing.

It would be very useful if there were a setting to force auto-generated SLA supports to be a certain minimum X/Y distance from the model throughout their body, probably excluding the top and bottom cones.

Occasionally I get one generating in contact with the model, which is not always easy to tell from the GUI when there is a forest of tightly-packed supports, and the support merges into the print making its removal more a matter of sculpting.

tamasmeszaros, 2019-02-15:
Hello,
the colliding supports issue is high priority and should be shipped in the final version, so don't worry :) 
As for an additional clearance parameter, it could be useful, we will consider your proposal.

Maeyanie, 2019-02-15:
I figured it might be a problem you were already planning to solve, but glad to hear that for certain.

bubnikv, 2019-02-17:
@Maeyanie By the way, what SLA printer are you printing on with our Slic3r?

Maeyanie, 2019-02-17:
@bubnikv I'm using a Wanhao Duplicator 7, with a NanoDLP controller.
Everything works pretty much out-of-the-box, and even in this early development stage, it's already better than both Wanhao and NanoDLP's slicers.
Aside from having make sure the slice thickness is the same in both Slic3r and NanoDLP (it doesn't read Slic3r's material settings, just the .png files) the only change I have to do is mirror the objects, which is quite trivial.
If there's anything you'd like me to test on it, feel free to ask, I'm always happy to help.

bubnikv, 2019-04-13:
Are you happy now with the support to object clearance? Could we close this issue?

Maeyanie, 2019-04-14:
Yes, it's pretty good now. I think in recent versions only once I've had something get too close and end up merging with the print. It was in a pretty bad location where there wasn't really a good place for Slic3r to put the support, so I don't blame it for that, and I just reoriented the model to give it space.

closed 2019-04-14 by Maeyanie