prusa3d/PrusaSlicer #1829: Alpha5 - "Save Project as" doesn't set default filename.

Sembazuru, 2019-02-16:
### Version
Version 1.42.0-alpha5-win64

### Operating system type + version
Windows 10 Pro - Version 1809

### Behavior
Something that I've noticed is when saving the project to a unique name with "SaveAs", the new name isn't preserved for either clicking the Save or generating gerbers. If I load 3 parts (say PartA.stl, PartB.stl, and PartC.stl) and then "SaveAs" as AllParts.3mf each time I try to save it will try to save as PartA.3mf. (Actually it could be PartA, PartB, or PartC. I haven't figured out how it decides which of the loaded stl filenames to use.) And when exporting g-code the gcode filename is based on one of the stl file names, not my project 3mf filename.

This is even more annoying after opening another project, and then saving as a new project name. Clicking save will overwrite the previous project forcing me to rebuild it.

This may be a symptom of the same issue as described in issue https://github.com/prusa3d/Slic3r/issues/1609

On the user forums I brought this up on the alpha5 thread and user marco.h6 confirmed the same behavior on MacOS.

Sembazuru, 2019-03-27:
This is still an issue on 1.42.0-beta-1

lukasmatena, 2019-08-26:
I believe this is no longer a problem in PrusaSlicer2.1.0. If there is a project open and you click 'export gcode' or 'Save project as...", the default name is always your project name. I'm closing the issue.

closed 2019-08-26 by lukasmatena