prusa3d/PrusaSlicer #1831: BUG: Unset PYTHONPATH, LD_LIBRARY_PATH, PERLLIB, etc before post processing

fusentasticus, 2019-02-16:
### Version
Slic3r 1.42.0-alpha5 (App Image)

### Operating system type + version
Linux 16.04

### Behavior
* Invoking a python script for postprocessing fails with bizarre complaints about modules not found
* Many other fail scenarios

### Work-around
In Python, use `-E` in shebang line as in 
```
#! /usr/bin/python3 -E
```
### Analysis
Inside the script, the line
```
sys.stderr.write("ENVIRONMENT:\n" +  "".join([ binding.__str__() + "\n" for binding in os.environ.items() ]))
```
reveals that the following environment variables are added or changed when the user's program is invoked from within the app image

('PYTHONHOME', '/tmp/.mount_Slic3rin96j0/usr/')
('XDG_DATA_DIRS', '/tmp/.mount_Slic3rin96j0/usr/share/:/usr/share/cinnamon:/usr/share/gnome:/usr/local/share:/usr/share:/var/lib/snapd/desktop:/var/lib/snapd/desktop:/usr/share/mdm/')
('LD_LIBRARY_PATH', '/tmp/.mount_Slic3rin96j0/usr/bin/bin')
('PYTHONPATH', '/tmp/.mount_Slic3rin96j0/usr/share/pyshared/:')
('PERLLIB', '/tmp/.mount_Slic3rin96j0/usr/share/perl5/:/tmp/.mount_Slic3rin96j0/usr/lib/perl5/:')

All such variables should be unchanged from the values given when Slic3r is invoked. In particular, most of them should remain unset!!!

Hywelmartin, 2019-02-20:
is this connected to why my script doesen't work?

```
#!/bin/bash

# replaces all
#
#  M204 S3500
#
#  with
#
#   ACC S=set value R= % of set value
#
# like this
#
#   ACC S=3500 R= 2450
while IFS='$\n' read -r line; do
  value=`echo "$line" | sed -rn 's/^M204 S([[:digit:]]+).*/\1/p'`
  if [ -z "$value" ];
  then
    echo $line
  else
    R_value=$((value * 70 / 100))
    echo "ACC S=${value} R=${R_value}"
  fi
done
```

fusentasticus, 2019-02-20:
I'd think it's rather my other issue #1832 that's the culprit -- you'll need to open for read the file that Slic3r gives you (in $1), then loop on all of it, then open $1 for write and stuff all output in there. Something like
```
echo $(while BLAH
...
done < "$1") > "$1"
```
or use `mktmp`

bubnikv, 2019-03-01:
@probonopd Please heeeelp!

For Slic3r 1.42.0-alphas, we are setting the LD_LIBRARY_PATH to point to the bundled libpng, that's all the path modifications we do in the slic3r starter batch script. All the rest is done by the Appimage magic.

What should we do to support starting of the Python or Perl post processing scripts from Slic3r packed into AppImage, so that the scripts are processed by the Python or Perl interpreter installed in the system?

probonopd, 2019-03-01:
Two thoughts:

1. Don't rely on Python or Perl being on the system - you can never know which Python or Perl may or may not be installed on the system. Rather bundle Python and/or Perl to have a "known good" one.
2. If you must call Python or Perl on the system: How are Python or Perl invoked by Slic3r? Can you unset those variables before launching Python or Perl?

probonopd, 2019-03-01:
Three thoughts:

1. Don't rely on Python or Perl being on the system - you can never know which Python or Perl may or may not be installed on the system. Rather bundle Python and/or Perl to have a "known good" one. Or are we talking about user-supplied scripts?
2. If you must call Python or Perl on the system: How are Python or Perl invoked by Slic3r? Can you unset those variables before launching Python or Perl?
3. If Slic3r PE does not use Python or Perl itself, then we don't need to set the `PYTHON*` and `PERL*` variables variables in the first place (by changing `AppRun`)

bubnikv, 2019-03-01:
1 is not an option, as we do not want to make the installation package huge for a fraction of Slic3r users.

2 is not a good solution as the user may have set those variables himself on the system.

3 is a good solution, but changing AppRun is out of our reach, at least in short term. I suppose AppRun is part of the AppImage magic. Why should AppImage rewrite the Perl & Python environment by default?

probonopd, 2019-03-01:
`AppRun` in the Slic3r PE AppImage is a bash script actually... check it out

probonopd, 2019-03-01:
`AppRun` in the Slic3r PE AppImage is a bash script actually... check it out. (It seems to be missing from this repo though?)

```
me@host:~$ /home/me/Downloads/Slic3rPE-1.42.0-alpha6+238-vk-octoprint-linux64-full-g8fd300c-201903011514.AppImage --appimage-extract AppRun
squashfs-root/AppRun
me@host:~$ cat squashfs-root/AppRun 
#!/bin/sh
HERE="$(dirname "$(readlink -f "${0}")")"
export UNION_PRELOAD="${HERE}"
export LD_PRELOAD="${HERE}/libunionpreload.so"
export PATH="${HERE}"/usr/bin/:"${HERE}"/usr/sbin/:"${HERE}"/usr/games/:"${HERE}"/bin/:"${HERE}"/sbin/:"${PATH}"
export LD_LIBRARY_PATH="${HERE}"/usr/lib/:"${HERE}"/usr/lib/i386-linux-gnu/:"${HERE}"/usr/lib/x86_64-linux-gnu/:"${HERE}"/usr/lib32/:"${HERE}"/usr/lib64/:"${HERE}"/lib/:"${HERE}"/lib/i386-linux-gnu/:"${HERE}"/lib/x86_64-linux-gnu/:"${HERE}"/lib32/:"${HERE}"/lib64/:"${LD_LIBRARY_PATH}"
export PYTHONPATH="${HERE}"/usr/share/pyshared/:"${PYTHONPATH}"
export PYTHONHOME="${HERE}"/usr/
export XDG_DATA_DIRS="${HERE}"/usr/share/:"${XDG_DATA_DIRS}"
export PERLLIB="${HERE}"/usr/share/perl5/:"${HERE}"/usr/lib/perl5/:"${PERLLIB}"
export GSETTINGS_SCHEMA_DIR="${HERE}"/usr/share/glib-2.0/schemas/:"${GSETTINGS_SCHEMA_DIR}"
export QT_PLUGIN_PATH="${HERE}"/usr/lib/qt4/plugins/:"${HERE}"/usr/lib/i386-linux-gnu/qt4/plugins/:"${HERE}"/usr/lib/x86_64-linux-gnu/qt4/plugins/:"${HERE}"/usr/lib32/qt4/plugins/:"${HERE}"/usr/lib64/qt4/plugins/:"${HERE}"/usr/lib/qt5/plugins/:"${HERE}"/usr/lib/i386-linux-gnu/qt5/plugins/:"${HERE}"/usr/lib/x86_64-linux-gnu/qt5/plugins/:"${HERE}"/usr/lib32/qt5/plugins/:"${HERE}"/usr/lib64/qt5/plugins/:"${QT_PLUGIN_PATH}"
EXEC=$(grep -e '^Exec=.*' "${HERE}"/*.desktop | head -n 1 | cut -d "=" -f 2- | sed -e 's|%.||g')
exec ${EXEC} "$@"
```

bubnikv, 2019-03-01:
Thanks. No more computer today.

vojtechkral, 2019-03-14:
This should be fixed now, a custom `AppRun` is supplied, essentialy the same one with some of the variables removed. In particular, the ones influencing Python and Perl are gone. `LD_LIBRARY_PATH` stays, as that's needed by AppImage I believe. It should not get in the way of Python scripts though.

closed 2019-03-14 by vojtechkral

probonopd, 2019-03-14:
If `LD_LIBRARY_PATH` is an issue, we can patch RPATHs instead using `patchelf`.

bubnikv, 2019-03-14:
> If LD_LIBRARY_PATH is an issue, we can patch RPATHs instead using patchelf.

Cool.