prusa3d/PrusaSlicer #1832: [Enhancement] Postprocessing -- allow script parameters, ban in-situ overwrite

fusentasticus, 2019-02-16:
I'm working on a refined Z-hop mechanism in a post processing script, but its function is of course dependent on a couple of parameters. I'd like to specify their values in the printer profile of course - as part of my script invocation in the `Output options` section:
```
/home/fusentasticus/fantastic_hop -mode awsome -height .3 
```
That doesn't work because the whole string is parsed as a filepath. I'd expect that the filename is just appended as the last item to arguments instead. Also, here, it must be noted: the current convention that the script modifies a file in-place is horrible in my gentle opinion; Unix orthodoxy calls for stdin to stdout processing and Windows allows that too, right?

bubnikv, 2019-03-01:
> That doesn't work because the whole string is parsed as a filepath. I'd expect that the filename is just appended as the last item to arguments instead.

I hope I already fixed that on Windows, it is left to be fixed on Linux, see
src/libslic3r/GCode/PostProcessor.cpp
functions quote_argv_winapi() and execute_process_winapi()
The functions above are using Win32 API to quote / unquote command line parameters.

We need to do something similar for Linux / OSX.

> Also, here, it must be noted: the current convention that the script modifies a file in-place is horrible in my gentle opinion; Unix orthodoxy calls for stdin to stdout processing and Windows allows that too, right?

You are completely correct. We just kept it the way we inherited it from upstream Slic3r.

vojtechkral, 2019-03-14:
This should be fixed in 4e510dc3e71d3eb96045c90cdf944396d0ca7f61

> Also, here, it must be noted: the current convention that the script modifies a file in-place is horrible in my gentle opinion; Unix orthodoxy calls for stdin to stdout processing and Windows allows that too, right?

It could be done but we prefer not to as it would break compatibility of existing scripts. Also, in-place file modification isn't entirely off-limits on Unix, consider for example `sed -i` ...

closed 2019-03-14 by vojtechkral