prusa3d/PrusaSlicer #1833: [Feature Request] Eco-Friendly Cut Skirt

jonnieZG, 2019-02-16:
Call me a tree-huger, but this is something that's been worrying me for quite some time. So here's my idea:

### Version
Any

### Operating system type + version
Applicable to all

### 3D printer brand / version + firmware version (if known)
Prusa MK3 (applicable to all)

### Behavior
* Currently the skirt is printed as a closed loop. Since more and more of the 3D-printed trash/remains end up in the environment, these loops/rings may pose risk of tangling for wild and marine animals.

### Proposed behavior
* Instead of printing the skirt as a closed loop, it could be printed with a tiny gap, so when removed from the bed it would be a string, rather than a closed loop.

Hywelmartin, 2019-02-22:
The best solution would be to not print it at all.. there's even a setting for that..

jonnieZG, 2019-02-22:
The purpose of a skirt is to stage the nozzle, so turning it off is a bad idea.

Hywelmartin, 2019-02-22:
So if you get a closed skirt around your print then the skirt is useless(if you only go one lap)... Prusa:s priming line makes most of the skirt useless 
The ability to have less than one lap would be good
Min/max length settings today there is only min and laps(integer)

ErikRL, 2019-02-22:
you could also cut the skirt when you take it off the bed.... as a solution in the meantime, if its something that is worrying you so much for so long... 

like mentioned the purge line "stages the nozzle" so turning it off isnt a bad idea. 
the couple line skirt can serve to fine tune any leveling flaws, so it can still serve a purpose outside of nozzle staging. 

the need or not, the want or not, really dont address the concern...

jonnieZG, 2019-02-26:
> you could also cut the skirt when you take it off the bed.... as a solution in the meantime, if its something that is worrying you so much for so long...

In fact, I have already been cutting the loop manually for the past year, but that's just me and perhaps few others. Adding that as a feature in Slic3r should motivate the others to consider the same. I like the skirt, because that's a nice chance to fine tune the Z-height, so I am always leaving it on.

bubnikv, 2019-03-01:
@jonnieZG I thought you are joking. Are you talking about your pets, or about the animals roaming around the land fill?

jonnieZG, 2019-03-01:
> @jonnieZG I thought you are joking. Are you talking about your pets, or about the animals roaming around the land fill?

With all the printers around generating many prints a day, it results in a significant amount of junk, and some of it ends up in the nature. Printing is a hobby that I really enjoy, but I became aware that it has an increasing effect on the environment. Since I am not willing to give it up, I thought this small and easy to implement thing might minimize the damage. I am not an ecologist, and I haven't done any science studies on this particular subject, but this is inspired by my logical conclusions.

riskable, 2019-03-05:
This is actually a good idea.  For a 0.4mm nozzle with PLA (or even ABS or PETG) the skirt is probably too thin to worry about it but I printed polycarbonate a while back with a 0.8mm nozzle as a test and that skirt was *ridiculously strong*!  It hurt my fingers to break it!

If it hurt my fingers to break the skirt (which was set to 3 loops BTW) then it's definitely possible for it to get stuck on an animal for what is essentially that animal's entire life.  I've seen turtles that developed with six pack rings stuck around them at our local Museum of Science and History up close.  I've also seen a seagull flying around with one stuck on its head/neck.

Yeah, it's simple enough to cut/break them but if we can solve it with a simple software change we should do that.  Even if it's only to save our fingers from the strength polycarbonate!

KnatteAnka, 2019-03-17:
an alternative is to use filament that can be recycled ;) then the loop don't matter and no plastic in nature ;) 
only PLA recycle exist in Sweden with Addnorth thou.

jonnieZG, 2019-03-18:
> an alternative is to use filament that can be recycled ;) then the loop don't matter and no plastic in nature ;)
> only PLA recycle exist in Sweden with Addnorth thou.

Good for you. If we continue the discussion in that direction, the cleanest thing would be to give up printing completely, but that is not constructive. Until the rest of the World bans non-eco-non-recyclable materials, it would be smart to do as much as we do to minimize the damage. This simple feature would be a small, but still a step forward.

sgoll, 2019-03-18:
I concur with @jonnieZG. This is one of those things where changing something just a little has the potential to avoid negative consequences for many others—in this case, animals that may be less affected by the plastic waste we produce as humanity.

Granted, it would be extremely difficult to support this claim scientifically (counting the number of plastic skirts out in the wild and checking how, if at all, wildlife is affected) but this still seems like a good idea. The print nozzle will be primed just as good with the gap.

Hywelmartin, 2019-04-01:
@jonnieZG If you only print useless things and toss them in the wild then you should quit printing immediately. Do you on the other hand print spare parts to extend life of existing products or anything else that is useful and solve any problem, then don't stop... For all waste is recycling the second best option, hiding things in landfills will never be a solution... to generate energy of bio based waste material is a effective way to utilize the waste.  But the best option is to reuse, but the ultimate is that the thing never was produced..

All materials for FDM is recyclable in some degree..

kmshort, 2019-05-15:
This is a good idea.

JanTuts, 2020-10-24:
I feel like some people here don't understand that, even if you were to exclusively use recyclable plastics, and neatly throw it in the recycling bin, it could **still** end up drifting in the ocean somewhere, get wrapped around some form of sea life, and choke it. 
People aren't perfect, and our garbage/recycling programs definitely aren't either.

Case in point: even when PLA _could_ be separated and recycled, it often **isn't** recycled by many recycling programs world-wide! 
In most places, it has the equivalent of Resin Identification Code 7 (Other), aka: _hard to / not recyclable => dump in a land fill, or burn in a furnace_
Also, if it doesn't get separated automatically (by floating, etc), and there's no code to be found on an item during manual separation (and those workers really aren't given more time than a quick glance at that conveyer they're stationed at), it automatically gets treated as "other" anyway...
(I've actually started experimenting with embossing the correct codes on parts that I print for this very reason)

And even if OP and many others consistently cut _their_ skirts before throwing them in the recycling bin, or even keep all their print trash neatly sorted at home and deliver it to a specialized recycling center... _thousands_ of other users aren't! 
Making the default skirt "pre-cut" (or at the very least weakened) right out of the slicer _automatically_ takes a step in solving the "potentially choking animals with your trash" issue for _everyone_... And of course this won't solve the _whole_ issue, but every step in the right direction counts.

And of course you could simply choose to _not_ use a skirt (or even make "no skirt" the default option), but if that then causes people's prints to come out wrong or utterly fail, requiring them to print another, then you're creating _even more_ trash!

kmshort, 2020-10-25:
Good idea, would be a nice option.

closed 2026-09-29 by prusaslicer-bot[bot]