prusa3d/PrusaSlicer #1837: 3D model of bed is removed with imported profiles

dsk001, 2019-02-17:
### Version
_Version of Slic3r Prusa Edition used goes here_
1.42.0-alpha5+win64
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 7 and 10
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa I3 MK3
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

If you import a printer config from a previous version of slic3r the 3d model of the bed is replaced with a grey grid plane. (see screenshot)
Even if you switch back in the pull down to the original profile provided with slic3r it doesnt change back until you restart the software.

_Is this a new feature request?_
no
#### Project File (.3MF)

 where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

[image]

bubnikv, 2019-03-01:
Slic3r PE paints the Prusa specific print beds for your custom print profiles only if it detects that they are derived from a valid prusa system profile.

I suppose the reason is, that we have renamed some of the printer profiles. The system profiles are referenced by name from your user profiles, so the link is now broken. You may fix it by manually editing the "inherits" field inside your custom profile file.

dsk001, 2019-03-01:
> Slic3r PE paints the Prusa specific print beds for your custom print profiles only if it detects that they are derived from a valid prusa system profile.
>
> I suppose the reason is, that we have [...]

Thank you for that information, it will be useful next time that might happen. Unfortunately I had already redone all my profiles manually (they were only 5).

closed 2019-04-13 by bubnikv