prusa3d/PrusaSlicer #1839: Extra solid layers when printing cylindrical shape

ultramango, 2019-02-17:
### Version
1.41.2

### Operating system type + version
Linux, Arch Linux, 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Object contains cylindrical cut-outs (laying "flat"). Slicing creates extra few solid layers in the bottom part of the cylinder cut-out (see screenshot for a see-trough) - those should not be there. Few initial layers are OK (cylindrical shape is visible), then few solid, then, until the end, it's OK.
Tried this workarounds but doesn't help:
- different printer profiles, layer heights, infill
- different imported 3D object formats
- insert extra tiny object where problem appears
This workaround works:
- rotate object and slice - extra layers are not there

#### Project File (.3MF) where problem occurs
[case-half_0.15mm_PLA_MK3.zip](https://github.com/prusa3d/Slic3r/files/2873161/case-half_0.15mm_PLA_MK3.zip)

### Screenshots
[image]

[image]

supermerill, 2019-02-19:
Maybe a problem with the stl? How many errors slic3r corrected?

ultramango, 2019-02-19:
> Maybe a problem with the stl?

Unlikely, in the end I sliced it in Cura and this artefact was not present. I used different formats (AMF, OBJ) and problem still persists.
I used FreeCAD to create the object, there wasn't anything there weird either.

> How many errors slic3r corrected?

Zero.

I also tested 1.41.3 using AppImage build from GitHub - same behaviour.

supermerill, 2019-02-19:
Can reproduce with 1.42.0-alpha5, 1.41.2

cannot reproduce with mainline slic3r 1.3.0

ultramango, 2019-02-19:
I did a go with a simple box + cylinder cutout (bool diff) - it slices correctly. I guess it must be a combination of few shapes that induce this bug.

bubnikv, 2019-03-11:
There seem to be a hole or flipped faces in your object.

[image]

If I let the model be fixed with the Netfabb service, it slices correctly.

> Maybe a bug from the commits with "Improved slicing of objects with cracks and gaps."

I would not call it a bug. For models with errors, different software packages use different heuristics to cope with the errors. Changing the heuristics for one set of errors may make the error handling worse for other models.

We invested a lot of time and money into making the slicing more robust for broken models. We keep a database of broken models, and every time we change the heuristics we re-test over our database. I know Cura works very well in this regard, our Slic3r is much better than Simplify3D when handling broken models.

I will keep this issue open, so one day we can return to it and possibly improve the heuristics for fixing slices over broken parts of the model.

closed 2026-09-29 by prusaslicer-bot[bot]