prusa3d/PrusaSlicer #1840: Unable to l

miko671995, 2019-02-18:
### Version
Slic3rPE-1.42.0-alpha5+win64-full-201902041815

### Operating system type + version
Win 10 64bit 1806 /newest update/

### 3D printer brand / version + firmware version (if known)
Ender 3 - custom profile + using firmware marlin / edited th3d R2.7 but my problem is to unable to slice /export gcode!)

### Behavior
Hello. Im newbie to your slicker and always tying new version to help you debug the best slicer. My problem depends only on one model /every another are slicable and exportable !) 
Only this rose is unable to slice. I runned my slicker exporting for 4 hours and nothing. After clicking on slicker it falls without any information of crash.

_Is this a new feature request?_
i will put here some fille but unable to export project file 

[image]

miko671995, 2019-02-18:
[Roses.zip](https://github.com/prusa3d/Slic3r/files/2874905/Roses.zip)

bubnikv, 2019-03-01:
It works for me.

Please attach a project file (a 3MF file) containing all your profiles, so we can reproduce your issue. Thanks.

miko671995, 2019-03-07:
> ** Fri, 01 Mar 2019 12:54:25 -0800 - "prusa3d/Slic3r" <slic3r@noreply.github.com>, "prusa3d/Slic3r" <reply@reply.github.com> **

Hello. I´m unable to make 3mf file because aplication crash. I use stock profiles https://github.com/sn4k3/Ender3 from this page and cureccury it is ender 3 STOCK. Harder models are slicing longer but it export some files but in this stl im not able to make files everytime app crashed .

bubnikv, 2019-03-11:
Then please export the config file by File->Export->Export Config
This way I will be able to reproduce your case exactly.

bubnikv, 2019-04-13:
Config was not provided after a month, closing.

closed 2019-04-13 by bubnikv