prusa3d/PrusaSlicer #1841: Improve object repairing from printabrick.org

gsedej, 2019-02-18:
### Version
1.42-0alpha5

### Operating system type + version
Linux (Ubuntu 18.04)

### Behavior
The _"Lego"_ bricks obtained from webpage http://printabrick.org/ are problematic to slice. The rendered model (non-sliced) looks OK, but after slicing, models breaks. Sometimes very much.
The below are just two examples. Many bricks (stl files) have many issues printing.

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_
the 3mf upload does not work (using drag an drop)...
The parts are in set https://printabrick.org/sets/8225-1
The number of part is 2994 and 6578
https://printabrick.org/bricks/2994
https://printabrick.org/bricks/6578
_The printabrick website sometimes isn't accasable._ 

Screenshots
correctly rendered wheel
[image]

missing axel holders after slicing
[image]

should be hole 
[image]

Tyre - redered correctly
[image]

problematic slice - should be hole, but it's filled
[image]

BerndJM, 2019-02-19:
Just for the records it's the same "problem" in Silc3r 1.41.3

To be honest, I do not think it's fair to blame a slicer when a STL file is totally broken.
Slic3r itself already fixes 888 errors! in the wheel.

I myself have been using Netfabb-Basic for a long time to repair broken STL's with a 99% success rate. But even Netfabb-Basic fails to repair this wheel terrific!

bubnikv, 2019-02-19:
@gsedej I wonder whether you shouldn't issue a bug at printabrick.org

gsedej, 2019-02-19:
I agree with both comments - the stl files are broken.
Use of Netfabb is suggested, but Netfabb stopped supporing linux. I know there are also some online tools, but either for pay or at least they need legitimate logoin main. 

Also I don't know how much can printprick.org do a help. They use LDraw library for models. They have github, but for the webpage. Maybe one could also report models.

I ask slice3r (PE) developers if they plan to improve fixing broken stl-s.

Slicing also depends on orientation. If I rotate object, I can get hole, but still no axel holder. Also much more errors (which can be somehow fixed with "detect thin walls" option)

[image]
(the fix with "detect thin walls" and "detect pridging primeters")
[image]

Using Cura 3.6, sliceing is done better.
[image]

But also broken, when rotating to preffered printing position.
[image]

BerndJM, 2019-02-19:
>  but Netfabb stopped supporing linux

Netfabb-Basic for Linux is still available for download up to version 7.4
https://github.com/3DprintFIT/netfabb-basic-download/releases

gsedej, 2019-02-21:
Thanks for suggestion. I did try fixing with netfabb, but even it didn't solve. 
_(it did help me with other parts, and I didn't have to give my phone number like online solutions)_

[image]

**So I ask the developers if they can improve the repair algorithm.**
If they don't plan to investigate broken models from printabrick.org, the bug can be closed (as wontfix or not-a-bug, etc)

bubnikv, 2019-02-21:
One of the models has randomly flipped triangles:

[image]

MisterPrintf, 2019-02-27:
Don't even bother. Printabrick is ethically wrong, and, in some cases, copyright violation. Give LEGO their money!

gsedej, 2019-02-28:
@MisterPrintf  You mean buying single/few bricks from 20 years ago? It's not like I am printing full brick set and potentially selling it. IIRC LEGO lost it's patent since there are many (chinese) companies selling full sets. Why would I hurt LEGO by being able to print out missing/broken/eaten pieces for my old collections?

On the topic: is there any Slic3r or Slic3r-PE developer that can comment on improving printaprick.org models stl slicing? Should I leave this issue opened?

MisterPrintf, 2019-02-28:
Honestly, Tinkercad's algorithm fails on most printabrick models, so I woudn't even touch it.

@gsedej printabrick's models are taken directly from ldraw which does not permit high-poly shapes so you are better off designing your own models.

bubnikv, 2021-03-12:
printabrick dead, the models are no more available.
Closing.

closed 2021-03-12 by bubnikv