prusa3d/PrusaSlicer #1842: Part of first layer missing

mdjurfeldt, 2019-02-18:
### Version
This applies both to the 1.41.3 release as well as version_1.42.0-alpha6-19-g644794233.

### Operating system type + version
I use Debian Stretch with some packages pulled in from unstable.

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* When slicing the Venus de Milo STL in the archive [venus.zip](https://github.com/prusa3d/Slic3r/files/2877423/venus.zip) a large section is missing in the first layer.
* Slice and look to reproduce!
* Expected Results: First layer should contain most of the area within the skirt.
* Actual Results: A large portion is missing as can be seen [here](https://user-images.githubusercontent.com/3612425/52977221-c308f580-33cc-11e9-828a-8bde075ed6f9.png).

BerndJM, 2019-02-19:
Nope, this part didn't lay absolutely flat on the printbed.
Try the "lay-flat" Button in the 3D-view and all will be fine ...
[image]

mdjurfeldt, 2019-02-19:
Thanks, I should have tried that myself.

However, let me tell you how I *did* check the flatness (since I expected something like this): I cut the model at 0.2 mm and could see that the model filled the first layer. In my mind, if a voxel is filled to more than 50%, the slicer should try to fill it with plastic. So, I guess I then post that as a feature request. :-)

bubnikv, 2019-03-01:
> Try the "lay-flat" Button in the 3D-view and all will be fine ...

You are right.

> In my mind, if local space between z_i and z_i+1is filled to more than 50%, the slicer should try to fill it with plastic.

Slic3r slices a layer at its center, that is at 0.1mm in your case. Half of your head model starts above 0.1mm. So in this case your request was fulfilled.

bubnikv, 2019-03-11:
Works as intended, closing.

closed 2019-03-11 by bubnikv