prusa3d/PrusaSlicer #1843: Wipetower m900 / m907 Wrong Code for reprap

Nappen, 2019-02-18:
### Version
1.41.3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Duet3d
Duet wifi 2.02(RTOS) (2018-12-24b1)

### Behavior
When running
Single extruder Multi Material 
and add Wipe tower.
it forced M900 and M907 command inside the .gcode files.
and reprap halt on them because both codes are unsupported.

In Reprap Firmware.
M900 is M572 D0 S0.1
M907 is M906

is it possible to Change this as slic3r Pe is a powerfull slic3r buth unusable if u are running dual extruder. dual on single hotend.

lukasmatena, 2019-04-24:
@Nappen The behavior was hopefully corrected by https://github.com/prusa3d/Slic3r/commit/08cb5bc2c7e57424d8783ca4aa02626dd0a1a48a, and it will be part of the next release. Please, let us know whether that solved your issues and none remain.

Some more tweaks might turn out to be needed (https://github.com/prusa3d/Slic3r/issues/1542), but I cannot tell when we have time to do them.

bubnikv, 2019-05-14:
It will be part of the PrusaSlicer 2.0.0-rc. Closing.

Please feel free to reopen if the above mentioned fix was not sufficient.

commit 08cb5bc2 (2019-04-24): Wipe tower uses correct gcodes for RepRap gcode flavor (M907->M906, M900->M572 - should fix #1843)
Also, fixed proper setting of the extruder current during toolchange (was broken since 6da83c7)
closed 2019-05-14 by bubnikv