prusa3d/PrusaSlicer #1844: Unexpected Infill tool path

BirkBinnard, 2019-02-19:
### Version
_Version of Slic3r Prusa Edition used goes here_

1.42.0-alpha5

### Operating system type + version

Win10-64 Home

### Behavior

Infill toolpath follows perimeter toolpath some of the time, but not always. Shouldn't it never follow the perimeter toolpath?

[image]

supermerill, 2019-02-19:
It works as intended. You can create an infill without "connection" between the lines, but these connections allow for a more constant-flow extrusion, less retraction, less z-hop, etc. 

Note that it's possible to add a setting to disable these lines (cf my fork), but it's only really useful for crating aesthetic stuff without perimeters.

BirkBinnard, 2019-02-19:
Very interesting - thanks.  Not what I'd call intuitively obvious by any means, but the I've never tried to understand the logic behind slicing algorithms.

bubnikv, 2019-03-01:
So can we close this issue?

bubnikv, 2019-03-11:
The perimeter connecting lines were explained. Closing.

closed 2019-03-11 by bubnikv