prusa3d/PrusaSlicer #1847: [Master 478032a Win64] Dialog box textboxes incorrectly sized

nebbywan, 2019-02-21:
### Version
Master branch commit 478032a

### Operating system type + version
Windows 10 v1809 Build 17763.316

### Behavior
Dialog box textboxes seem to be incorrectly sized. The issue is more apparent on a 1080p monitor, but is still present on a HiDPI display.

1080p:
[image]

HiDPI (4K):
[image]

While the examples are of the "Send to printer" dialog, I also noticed this on the "configuration update" dialog as well.

bubnikv, 2019-02-21:
Great that you took time to review  HiDPI support at the current master. Please report the other flaws in HiDPI support if you stumble over them. We hope the OctoPrint integration dialogs are the only remaining cases, but we may have overlooked some cases yet.

@vojtechkral please adjust the OctoPrint integration dialogs for HiDPI.

nebbywan, 2019-02-22:
Looking perfect on my 1080p machine with commit 14b4685. Unfortunately my HiDPI machine is out for repair due to my own klutzy nature, so I won't be able to confirm anything on that side until (probably) Tuesday, but I'll close this issue anyhow and just reopen if necessary. Thanks for the quick fix!

closed 2019-02-22 by nebbywan