prusa3d/PrusaSlicer #1848: Modifier causes shutdown

skenecroft, 2019-02-21:
1.41.3+Version
_

Latest version of OSX 10.14.2

Prusa Mk3

I have a lid for an electronics box and i want to add some ventilation. One way todo this is to not print the top and bottom layers as then you just get the infill which is holey.

If i print the entire lid with no top or bottom layers it will look very odd so instead i want to create a modifier mesh in the shape of a cylinder where there is no top or bottom layers

When i set the top or bottom layers to zero the program crashes

bubnikv, 2019-03-01:
> When i set the top or bottom layers to zero the program crashes

It crashes first when you start slicing, right? Or do you have background processing enabled?

Do you have the same issue with 
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha5
?

skenecroft, 2019-03-14:
The crash happens when i use a modifier to remove a bottom layer to remove the bottom layers

It crashes the moment i say OK before i go near slicing

I got a crash report last time it went tits up

Process:               perl5.26.1 [43615]
Path:                  /Volumes/VOLUME/*/Slic3r.app/Contents/MacOS/perl5.26.1
Identifier:            com.prusa3d.slic3r/
Version:               Slic3r 1.41.3+ (1.41.3+)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           perl5.26.1 [43615]
User ID:               501

Date/Time:             2019-03-14 19:54:10.034 +0000
OS Version:            Mac OS X 10.14.2 (18C54)
Report Version:        12
Anonymous UUID:        5604CA5C-833F-AB32-1C30-4AD4E4C3FF27

Time Awake Since Boot: 260000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [43615]

VM Regions Near 0:
--> 
    __TEXT                 000000010c75d000-000000010c91d000 [ 1792K] r-x/rwx SM=COW  /var/folders/*/Slic3r.app/Contents/MacOS/perl5.26.1

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   ???                           	000000000000000000 0 + 0
1   Wx.bundle                     	0x000000010e2619eb XS_Wx__TreeCtrl_GetSelection(interpreter*, cv*) + 123
2   perl5.26.1                    	0x000000010c81f1fd Perl_pp_entersub + 2365
3   perl5.26.1                    	0x000000010c815cb6 Perl_runops_standard + 22
4   perl5.26.1                    	0x000000010c7860d1 Perl_call_sv + 1089
5   Wx.bundle                     	0x000000010e2a7216 wxPliEventCallback::Handler(wxEvent&) + 422
6   libwx_baseu-3.0.dylib         	0x000000010d7df2aa wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 186
7   libwx_baseu-3.0.dylib         	0x000000010d7df01b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
8   libwx_baseu-3.0.dylib         	0x000000010d7def2e wxEvtHandler::ProcessEvent(wxEvent&) + 190
9   libwx_baseu-3.0.dylib         	0x000000010d7df34f wxEvtHandler::SafelyProcessEvent(wxEvent&) + 15
10  libwx_osx_cocoau_core-3.0.dylib	0x000000010daf9923 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 67
11  libwx_osx_cocoau_core-3.0.dylib	0x000000010d9fe676 wxAppBase::ProcessIdle() + 134
12  libwx_osx_cocoau_core-3.0.dylib	0x000000010d970eaa wxApp::ProcessIdle() + 26
13  libwx_baseu-3.0.dylib         	0x000000010d7006cc wxEventLoopBase::ProcessIdle() + 28
14  libwx_baseu-3.0.dylib         	0x000000010d7b41f7 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 87
15  com.apple.CoreFoundation      	0x00007fff4c4409cd __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
16  com.apple.CoreFoundation      	0x00007fff4c440902 __CFRunLoopDoObservers + 452
17  com.apple.CoreFoundation      	0x00007fff4c3e2558 __CFRunLoopRun + 1469
18  com.apple.CoreFoundation      	0x00007fff4c3e1d48 CFRunLoopRunSpecific + 463
19  com.apple.HIToolbox           	0x00007fff4b678ab5 RunCurrentEventLoopInMode + 293
20  com.apple.HIToolbox           	0x00007fff4b6787eb ReceiveNextEventCommon + 618
21  com.apple.HIToolbox           	0x00007fff4b678568 _BlockUntilNextEventMatchingListInModeWithFilter + 64
22  com.apple.AppKit              	0x00007fff49933363 _DPSNextEvent + 997
23  com.apple.AppKit              	0x00007fff49932102 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1362
24  com.apple.AppKit              	0x00007fff4992c165 -[NSApplication run] + 699
25  libwx_osx_cocoau_core-3.0.dylib	0x000000010d9d845e wxGUIEventLoop::OSXDoRun() + 110
26  libwx_baseu-3.0.dylib         	0x000000010d7b4b14 wxCFEventLoop::DoRun() + 52
27  libwx_baseu-3.0.dylib         	0x000000010d700595 wxEventLoopBase::Run() + 165
28  libwx_baseu-3.0.dylib         	0x000000010d6c99c1 wxAppConsoleBase::MainLoop() + 209
[... 497 lines cut ...]
Model: iMac18,3, BootROM 166.0.0.0.0, 4 processors, Intel Core i5, 3.8 GHz, 8 GB, SMC 2.41f1
Graphics: Radeon Pro 580, Radeon Pro 580, PCIe
Memory Module: BANK 0/DIMM0, 4 GB, DDR4, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334532202020
Memory Module: BANK 1/DIMM0, 4 GB, DDR4, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334532202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.61.1 AirPortDriverBrcmNIC-1305.2)
Bluetooth: Version 6.0.9f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST3000DM001, 3 TB
USB Device: USB 3.0 Bus
USB Device: My Passport 0748
USB Device: External USB 3.0
USB Device: PTN5002
USB Device: PTN5002
USB Device: USB2.0 Hub
USB Device: SpaceNavigator
USB Device: USB2.0 Hub
USB Device: Hub
USB Device: USB2.0 HUB
USB Device: USB2.0 HUB
USB Device: USB Storage
USB Device: Apple Cinema Display
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 39.2

FidelCapo, 2019-03-15:
I am unable to reproduce this issue. Can you please attach 3mf file? Thank you.

skenecroft, 2019-03-15:
I have no idea what a 3mf is

but have a youtube video

https://youtu.be/WhRZ1bxHjdE

skenecroft, 2019-03-17:
I have now tried 1.42 and this issue is no more

And there was much rejoicing

bubnikv, 2019-03-17:
great, thanks for the note.

closed 2019-04-13 by bubnikv