prusa3d/PrusaSlicer #1850: Segmentation fault while slicing

mdjurfeldt, 2019-02-21:
### Version
1.41.3 as well as version_1.42.0-alpha6-19-g644794233

### Operating system type + version
Debian Stretch with some newer libraries

### 3D printer brand / version + firmware version (if known)
Stock Prusa MK3 + MMU2

### Behavior
* I get a segmentation fault when slicing a model.
* Add 
[this](https://github.com/prusa3d/Slic3r/files/2889484/camera-holder.zip) STL and rotated 90 degrees around the X axis. Then slice it with 0.2mm soluble interfaces. That should give you a segmentation fault. If not, please let me give more specific settings.

bubnikv, 2019-03-01:
I tested quickly with the current master, and I was not able to reproduce your crash.

Do you get the crash with the current latest alpha? 
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha5

We are now investing all our effort into releasing the next big thing.

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv

mdjurfeldt, 2019-04-13:
Sorry---I forgot to answer this. Did you see that I had tested it with 1.42.0-alpha6? Do you want me to test it with 1.42.0-beta1 now?

bubnikv, 2019-04-14:
yes, please

mdjurfeldt, 2019-04-17:
I am sorry, but now I can't reproduce the segmentation fault anylonger. It must be either something related to my system or a Heisenbug. I suggest that we close this issue.