prusa3d/PrusaSlicer #1852: Feature Request: Nozzle pre heat temperature to allow for not polluting the bed during bed leveing

rennne, 2019-02-21:
There are quite a lot Issues open which are discussion to how to change the start g-code in order to prevent oozing during mesh bed leveling.

It boils down to, that the nozzle has to be at a different temperature during bed leveling than during printing.

This temperature has to be at a point where the oozing is keep at a minimum, but also has to be high enough that any plastic which may be sticking on the nozzle is soft enough to not interfere with the mash bed leveling process.

I tried a few different materials and found:
Prusa PLA is best preheated to 205 but prints at 210
Prusa PETG is best preheated to 190 but prints at 240
PC/PBT should be preheated to 210 but prints at 240

For some filaments 10 degree more or less and it could be to hard or to soft.

The point is, that this value can not easy be derived from the print temperature, also one value for all does not work.

I think the most future prove approche would be to have a 
- New Field: Filament Settings -> Temperatures -> Extruder: -> Pre Heat

This could then be used in the G-Code before triggering G80 like it is discussed in some places:
[Slic3r-settings/issues/45](https://github.com/prusa3d/Slic3r-settings/issues/45)
[Prusa-Firmware/issues/699](https://github.com/prusa3d/Prusa-Firmware/issues/699)
[Prusa-Firmware/issues/1239](https://github.com/prusa3d/Prusa-Firmware/issues/1239)
[Slic3r/issues/1126](https://github.com/prusa3d/Slic3r/issues/1126)

tsprhdm, 2019-07-02:
I'd rather set to a low temperature and make it regular that before starting a job, the operatorneeds to: 

- clean the bed / plate
- make sure the nozzle(s) are clean in the first place.

Then, assuming a clean nozzle, with low enough temperature, there should not be any oozing at all, therefore, nothing sticking out from the nozzle durig mesh bed leveling!

guestisp, 2019-07-02:
Much easier to add 1 gcode line in the start gcode: M104 S150 before the G29 then after g29 park the nozzle: G0 X-5 Y-5

During the warm up the nozzle is outside the bed

tsprhdm, 2019-07-02:
I put the park at Z=50, so there can be a final nozzle clean with a wirebrush.

closed 2026-09-29 by prusaslicer-bot[bot]