prusa3d/PrusaSlicer #1853: Bug: Model rendering

Thoralf690, 2019-02-21:
### Version Slic3rPE-1.42.0-alpha5+win64-full-201902041815

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17134), 64-bit edition

OpenGL installation
GL version:   4.6.0 NVIDIA 417.35
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 960/PCIe/SSE2
GLSL version: 4.60 NVIDIA

Rendering goes all crazy (some triangles becom transparent and support enforcers become invisible) after the following steps:
1. Loading Model
2. Placing on Face (Cable guide is upright)
3. add support enforcer slab 
4. Switch to front view (Num 3)
5. Moving the support enforcer causes the rendering to crash

After checking the .3mf file in Meshmixer it might not just be a rendering issue but corruption of the .3mf file itself. Meshmixer loads the plate but only shows some triangles spread over the buildplate.

[Rendering Bug.zip](https://github.com/prusa3d/Slic3r/files/2890927/Rendering.Bug.zip)
[Model.zip](https://github.com/prusa3d/Slic3r/files/2890931/Model.zip)

enricoturri1966, 2019-02-22:
Working on it -> http://dev.prusa3d.com/browse/SPE-833

FidelCapo, 2019-03-01:
Thanks for reporting, this issue should be fixed in next release.
Closing.

closed 2019-03-01 by FidelCapo