prusa3d/PrusaSlicer #1854: Feature Request: "Only retract when crossing perimeters" to allow better results with "avoid crossing perimeters"

rennne, 2019-02-21:
Using the "avoid crossing perimeters" function can help a lot for some difficult prints or materials.

Depending on the situation it can be necessary to deactivate retraction and/or z-lift when doing so.
But when the nozzle has to cross the perimeter, retraction and/or z-lift could then be necessary again to archive good results.

Therefore two additional open in the Extruder settings would be useful:
- Only retract when crossing perimeters
- Only z-hop when crossing perimeters

dasquiddi, 2020-02-06:
i've searched recently for an option that do the "Only z-hop when crossing Perimeters" 
are there any improvements or tips do implement/activate this behavior?

bubnikv, 2020-12-11:
And there is a request to do just
https://github.com/prusa3d/PrusaSlicer/issues/2112

closed 2026-09-29 by prusaslicer-bot[bot]