prusa3d/PrusaSlicer #1855: Long Skinny Sections come out weak due to Automatic extrusion width calculation flaws.

Isaak-Malers, 2019-02-22:
### Version
1.41

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
Custom Delta
Kossel
Duplicator 6
Maker Select Plus
FT-5
N/A * all printers exhibit this issue.

### Behavior
Slic3r performs very poorly in printing long runs of a part that are barely too thin to require infill.  This kind of model is extremely common when trying to use a 3d print to emulate an injection molded part.  To reproduce issue, print the linked part with the following conditions:

* .4mm nozzle
* 3 perimeters
* Nylon, Cheap ABS, Or any other "tricky because its prone to under-extrude" filament.  Note that this kind of filament is not a requirement for this issue (given certain parts i can replicate with PLA), but it does make it easier to reproduce.
* Slicer settings on default (1 print)
* Slicer settings on default, but with .5mm extrusion widths over-ridden for everything. (1 print)

Test STL:
[printer-test.zip](https://github.com/prusa3d/Slic3r-settings/files/2892959/printer-test.zip)

Results: Perimeters do not bond together, and it is possible to "splinter" the print apart.  Depending on how many perimeters there are.  Note that this is not an issue with under-extrusion or printer "slop"  2/4 of the walls on this print come out great for me,  and the results are the same regardless of orientation of the part.

[image]

Expected Results:  The part shouldn't be so weak.  But more specifically I could see this being solved in 2 ways:

1. More sophisticated extrusion width/flow rate calculation, filling in the area with solid plastic.

2. Instead of printing a part with all the requested perimeters, reduce the number of perimeters and print the "inner perimeters" as solid infill instead.

bubnikv, 2019-02-22:
This sounds like an under-extrusion issue, and you need to calibrate your
extruder e-steps.

>  Instead of printing a part with all the requested perimeters, reduce the
number of perimeters and print the "inner perimeters" as solid infill
instead.

Do you have "Detect thin walls" enabled?

Isaak-Malers, 2019-02-22:
I've noted this problem on 5+ printers, all before and after estep calibration.  In the STL i linked certain parts of the model will print poorly and other will print well.  A hardware issue wouldn't come and go so reliably depending on wall thickness.

I do have "Detect thin walls" enabled, but I don't think that is relevant.  The thinnest wall on my test print is 1.5mm across.  My understanding is that Thin Walls are walls less than a single extrusion width.

Isaak-Malers, 2019-02-22:
Upon further investigation, this appears to be an issue caused by the fact that different fillaments have different cross sections.  A good fix would be allowing setting the extrusion multiplier individually for each of the extrusion width types.  Infill, first layer, perimiters, etc.  A more complex solution might be allowing users to tune the extrusion shape function.

Then for the "interior perimiters" I can set an extrusion multiplier that will fix my gapfill issue without un-calibrating the rest of my print.

Isaak-Malers, 2019-02-23:
After doing more research here are my findings:

The assumptions made on this page:

closed 2026-09-29 by prusaslicer-bot[bot]