prusa3d/PrusaSlicer #1856: Copy / paste keyboard shortcuts not working properly on macOS Mojave

PSchaug, 2019-02-22:
Slic3rPE 1.42.0 Alpha 5+
macOS Mojave 10.14.2

In some dialogue boxes macOS copy/paste shortcuts do not work.
To reproduce:
Press Export G-code... Trying to use cmd-A (select all, cmd-C (copy) or cmd-V (paste) does not give expected results, nothing happens. Need to use right-click and menu commands for this to work.

Same result in other dialogue boxes (basically any dialogue on the File-menu).

bubnikv, 2019-04-13:
Do you still have the issue with the beta1? I believe we updated to a newer wxWidgets UI library, so it may have improved.

Would you please list out all the UI elements, where you have this issue? Thanks.

PSchaug, 2019-04-13:
Hi, issue still exists in beta1.
It is easy to replicate:
In the main window, click on Send G-code. Press Cmd-X to cut text, text disappears. Press Cmd-V to paste, text is back.
Then, in the main window, click on Export G-code. Press Cmd-X, nothing happens. Cmd-V, nothing happens. This is inconsistent behaviour, I have no idea why.
I have found this in the following UI elements:
- Export G-code-button on Plater screen
- All UI elements in the File menu where there is a dialogue box

Any other info you need? I'm now running macOS Mojave 10.14.3, no difference from earlier.

bubnikv, 2019-04-13:
Thanks for the detailed report. It looks like a bug in the multi-platform UI library we are using (the wxWidgets).

DominikPalo, 2019-05-12:
I have the same issue - when I try to copy/paste a value of some setting (for example in the Filament Settings window) using keyboard shortcuts Cmd+C/Cmd+V, the currently loaded 3D model in the Plater tab is duplicated instead of the setting text value.

bubnikv, 2019-05-14:
@DominikPalo
> I have the same issue - when I try to copy/paste a value of some setting (for example in the Filament Settings window) using keyboard shortcuts Cmd+C/Cmd+V, the currently loaded 3D model in the Plater tab is duplicated instead of the setting text value.

This is actually a different issue than the original one. The one by @DominikPalo has been introduced with the copy / paste functionality in the 3D scene. The issue should be now fixed in the PrusaSlicer 2.0.0-rc.

Technofrikus, 2020-11-25:
This bug is still there. Tested on Mojave and Catalina, and on 2.2.0 and 2.3.0 Alpha 4.

Although right clicking and copy/pasting does work. Just the shortcuts do not work.

FidelCapo, 2020-11-25:
@Technofrikus I cannot reproduce this issue. I can copy/paste in tabs by the shortcuts.. Can you copy/paste anywhere in slicer?

Technofrikus, 2020-11-25:
In the Print Settings for example I can copy/paste with shortcuts. But as PSchaug mentioned, in the Export G-Code dialog (file name) I cant. Cmd + A (select all) doesnt work as well.

PSchaug, 2020-11-25:
The problem has been the same as long as I have used Slic3rPE / PrusaSlicer, through several OS upgrades. In any dialog box from the File-menu the keyboard shortcuts do not work (File/Save Project as, any itek on the File/Export-menu). As Technofrikus says, right-click brings up the correct menu to select copy/paste.
Currently using Norwegian keyboard layout, MacOS Mojave 10.14.6, English language.

FidelCapo, 2020-11-26:
Ok, thank you. I can reproduce this issue now.

bj97301, 2024-07-07:
any eta on this fix?  Love prusaSlicer btw.  you guys are amazing and finally saved up enough to get a used mk3s.  fantastic machine 👏

closed 2026-09-29 by prusaslicer-bot[bot]