prusa3d/PrusaSlicer #1857: [Feature Request] option for alignment holes on cut

Animo78, 2019-02-22:
Good day,
There is an option which I would like to see.
Which isn't in any slicer/slic3r I tried so far.

It's sometimes difficult to glue cut parts, Wich were to large for the printbed.
It would be awesome if there was an option to add a few holes (the size of the filament itself) after the cut.
So you can put in a tiny piece of filament in one part, then asd glue to the part, then put the other half on top, being alligned by the filament.

Thank you in advance.

supermerill, 2019-02-27:
similar issue in mainline: https://github.com/slic3r/Slic3r/issues/4709

James-G4CLF, 2019-12-24:
I do this myself when making an object in parts rather than using CUT.
I use 1.8 mm holes which I ream with a 1.8mm drill.

It would be very useful to add a feature allowing insertion of such holes across the cut plane before making the cut.

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena