prusa3d/PrusaSlicer #1858: Slic3r PE wont open

NightmareWokeUp, 2019-02-22:
### Version
Can't tell because uninstalled, but not older than 1month

### Operating system type + version
Windows 10

### Behavior
Worked fine until now. Exported Model from Fusion360 to SlicerPE like usual, then when changing the Printer settings the program suddently closed. Wasnt able to open it again, doesnt show up in task manager. Anti-virus not the problem either, really not sure what happened or why. Or how to reproduce it.

bubnikv, 2019-03-01:
Does slic3r start if you clear its configuration folder?

bubnikv, 2019-03-01:
Running Slic3r 1.41.3 from command line using your configuration, I am getting

D:\temp\Slic3rPE-1.41.2+win64-full-201811221508>slic3r-console.exe "--datadir=D:\temp\crashboombang\1858\slicer backup"
Creating main frame...
Error in function Slic3r::Flow::new_from_config_width at ..\xs\src\libslic3r\Flow.cpp:31: Invalid flow height supplied to new_from_config_width()
         at D:/temp/Slic3rPE-1.41.2+win64-full-201811221508/lib/Slic3r/GUI/MainFrame.pm line 191.
        Slic3r::GUI::MainFrame::_init_tabpanel(Slic3r::GUI::MainFrame=HASH(0x2d80ce823e0)) called at D:/temp/Slic3rPE-1.41.2+win64-full-201811221508/lib/Slic3r/GUI/MainFrame.pm line 56
        Slic3r::GUI::MainFrame::new("Slic3r::GUI::MainFrame", "no_controller", 1, "no_plater", undef, "lang_ch_event", 10320, "preferences_event", ...) called at D:/temp/Slic3rPE-1.41.2+win64-full-201811221508/lib/Slic3r/GUI.pm line 123
        Slic3r::GUI::OnInit(Slic3r::GUI=HASH(0x2d80aeb54c8)) called at D:/temp/Slic3rPE-1.41.2+win64-full-201811221508/lib/Wx/App.pm line 36
        eval {...} called at D:/temp/Slic3rPE-1.41.2+win64-full-201811221508/lib/Wx/App.pm line 36
        Wx::App::new("Slic3r::GUI") called at D:\temp\Slic3rPE-1.41.2+win64-full-201811221508\slic3r.pl line 114

Checking your print/Ender 3 fast.ini
there is
layer_height = 1.28828e-312

Changing it to 
layer_height = 0.2

Slic3r starts merrily.

Now I wonder how that happened.

NightmareWokeUp, 2019-03-01:
Hello, thanks for your reply!
Yeah I noticed that too when copying the old config to the fresh install.
Wonder too how that happened. Worked fine and I cant imagine Fusion opening
an .exe would change the config files?

Best regards

bubnikv, 2019-03-11:
> Worked fine and I cant imagine Fusion opening an .exe would change the config files?

that should not happen indeed.

I am going to close this issue. Let's hope it will not repeat with the 1.42.0 releases.

closed 2019-03-11 by bubnikv