prusa3d/PrusaSlicer #1859: Octoprint connection issue

svatos-jirka, 2019-02-23:
### Version
1.41.3+win64

Slic3r Prusa Edition
Version: 1.41.3+win64
Build:   Slic3rPE-1.41.3+win64-full-201902111723
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 6.1.7601]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\5.24.0
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5\MSWin32-x64-multi-thread
    C:\PROGRA~1\Prusa3D\Slic3rPE\local-lib\lib\perl5
    C:/PROGRA~1/Prusa3D/Slic3rPE/lib
    C:/Program Files/Prusa3D/Slic3rPE/lib
    .
OpenGL installation
GL version:   4.5.0 NVIDIA 354.59
Vendor:       NVIDIA Corporation
Renderer:     Quadro M2000M/PCIe/SSE2
GLSL version: 4.50 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
[... 286 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### MK3 Prusa

### Behavior
After upgrade to Slic3r PE 1.41.3 connection to octoprint Version 1.3.10 is very unstable. Sometime works, sometime it takes too long time. Also cancelation of "upload dialog" leads to slic3r PE freeze

vojtechkral, 2019-02-24:
Thanks for the report. FWIW I believe there was not change to Octoprint code between 1.41.1 and 1.41.3.

bubnikv, 2019-04-13:
Do you still have the issue? Do you have the issue with https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1 ?

closed 2019-04-13 by bubnikv