prusa3d/PrusaSlicer #1860: [FR] Counterbore hole "hymen"

radek-senfeld, 2019-02-23:
Sometimes there's a need to print counterbore holes oriented upside down.

I usually go countersink or use chamfered edges on my designs but sometimes you just download an unprepared model from Thingiverse and don't want to mess with it..

I'm aware using supports is the preferred way to deal with this but it would be really sweet to have an option to place one full bridged layer just beneath the size transition. It's really easy to perforate this "hymen" with the bolt afterwards.

I'm not sure how hard is this feature to implement but even a simple tool to manually mark such geometries would make a huge difference in usability and I presume I'm not the only one who would appreciate such a function.

I haven't tried using modifiers for this yet. I guess it could be done but I expect the positioning to be laborious.

Placing a modifier from Preview tab might help as you would clearly see what you are doing and have the Z height set at least.

Issue #127 seems quite close but I'm asking humbly and not expecting too much automagic. :-)

Spacemarine2018, 2019-02-25:
I think it's a great idea to manually insert bridges. Maybe these bridges could also be used as a starting plate for support?

supermerill, 2019-02-28:
Maybe you can create a cylinder part, with a height = your layer height, and place it at the right place?

bubnikv, 2019-03-01:
> Maybe you can create a cylinder part, with a height = your layer height, and place it at the right place?

Exactly. With Slic3r 1.42.0-alpha5 
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha5
this is pretty simple to do. Just add a generic cylinder part, set the scaling and position from the side bar and you are done.

radek-senfeld, 2019-03-05:
Adding a generic cylinder is what I meant by using modifiers. Tried to blind a hole once in 1.42.0-alpha4 and it was quite tedious to position and size the cylinder properly. Not even imaging a model with more holes in it..

Is there some new spell in 1.42.0-alpha5? I'll give it another shot and report back.

radek-senfeld, 2019-03-05:
Well, it can be done but the process is IMHO far from being "pretty simple to do".

 * Switch to Expert mode.
 * Ctrl + I -> Insert STL.
 * Right click on the model and select Add part -> Cylinder.
 * I end up with two parts in menu on the right.
 * When I click the cylinder, both the original part and the cylinder gets selected. I need to use the menu on the right to select just the cylinder.
 * Now the fun begins.. You move the cylinder.. Switching the Plate and Preview tabs until satisfied with the result.
 * Preview doesn't keep the slider set from previous iteration so you need to adjust it with every reslice.
 * Coordinate system has (0,0) in the middle of the plate. Do your math..

Btw: Del doesn't work when the part is selected via the menu on the right. I need to click Edit -> Delete selected. Is this an intended behavior or just a bug?

supermerill, 2019-03-05:
> Coordinate system has (0,0) in the middle of the plate. Do your math..

i think 0,0 is in the middle of the part. If you move the part in an other section of your bed, you don't have to modify that (i can be wrong).

dartrax, 2026-09-18:
Still a valid FR