prusa3d/PrusaSlicer #1861: Typo in Ukrainian translation of the menu 'Help'

omgovdan, 2019-02-23:
### Version
1.41.3+win64

### Operating system type + version
Windows10 x64

### Behavior
* Typo in Ukrainian translation of the menu 'Help'
* _Expected Results_: 'Допомога'
* _Actual Results_ : 'Доромога'
  * _Screenshots_ 
  
[image]

YuSanka, 2019-02-26:
Thanks for your report. This typo is fixed and will be a part of an upcoming release. But if you have a time and/or wish to help us with Ukrainian localisation, you are welcome :)

closed 2019-02-26 by bubnikv

omgovdan, 2019-02-26:
Sure. How can I do that?

Regards,

Alex

YuSanka, 2019-02-26:
O! It's very nice :)

So, for translation we use [PoEdit](<https://poedit.net/>)

Take [Slic3r_uk.po](<https://github.com/prusa3d/Slic3r/tree/master/resources/localization/uk>) from our site, open it with PoEdit and translate a missing phrases.
It's very important to keep attention to every gaps and punctuation. Especially with formated strings. (used %d, %s etc.) 

When you save a Slic3r_uk.po, Slic3r_uk.mo will be created automaticaly. Rename it to the Slic3r.mo and do pull-request

omgovdan, 2019-02-26:
Got it. Will try on weekend

Regards,

Alex