prusa3d/PrusaSlicer #1864: Zeroconf/Bonjour does not work in Slic3r PE AppImage

probonopd, 2019-02-24:
### Version

1.41.1 AppImage

### Operating system type + version

Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)

n/a

### Behavior
* Printer Settings -> Printer Host upload -> Browse...
* Does not recognize any printers advertised with Bonjour in the network

### Additional information

It looks like `libnet-bonjour-perl` and its dependencies are not bundled inside the AppImage, hence Zeroconf/Bonjour can not work in the Perl bundled inside the AppImage.

vojtechkral, 2019-02-25:
Can you see those printers in an Avahi Zeroconf browser or equivalent? Is there anything special about them or your network setup?

> It looks like `libnet-bonjour-perl` and its dependencies are not bundled inside the AppImage, hence Zeroconf/Bonjour can not work in the Perl bundled inside the AppImage.

That's because Perl bonjour implementation is not used since 1.40.0. I wrote one in C++.

bubnikv, 2019-02-25:
By the way, the Slic3r 1.42.0-alpha5 is the current latest development
release, and it is perl free. Please test whether you have an issue with
the alpha5. The Slic3r 1.41.3 will likely not be updated.

probonopd, 2019-02-25:
I need to correct myself:

> Does not recognize any printers advertised with Bonjour in the network

Actually it __does__ recognize an original OctoPrint instance, but it __does not__ recognize my self-made OctoPrint-like [ESP8266 WirelessPrinting](https://github.com/probonopd/WirelessPrinting) solution.

This is the original OctoPrint, and it __is recognized by Slic3r PE__:

[image]

```
me@host:~$ avahi-browse -ar

(...)
= enp0s25 IPv4 OctoPrint instance on host                    _octoprint._tcp      local
   hostname = [host.local]
   address = [192.168.0.15]
   port = [5000]
   txt = ["version=1.3.10" "api=0.1" "path=/"]
(...)
```

This is it in Wireshark:

```
0000   00 00 03 04 00 06 00 00 00 00 00 00 00 00 08 00   ................
0010   45 00 00 f7 ef fe 40 00 ff 11 09 88 c0 a8 00 0f   E..÷ïþ@.ÿ...À¨..
0020   c0 a8 00 0f 14 e9 e2 a8 00 e3 82 63 20 81 84 00   À¨...éâ¨.ã.c ...
0030   00 01 00 06 00 00 00 00 0a 5f 6f 63 74 6f 70 72   ........._octopr
0040   69 6e 74 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00   int._tcp.local..
0050   0c 00 ff c0 0c 00 0c 00 01 00 00 00 0a 00 1d 1a   ..ÿÀ............
0060   4f 63 74 6f 50 72 69 6e 74 20 69 6e 73 74 61 6e   OctoPrint instan
0070   63 65 20 6f 6e 20 68 6f 73 74 c0 0c c0 33 00 10   ce on hostÀ.À3..
0080   00 01 00 00 00 0a 00 1e 06 70 61 74 68 3d 2f 07   .........path=/.
0090   61 70 69 3d 30 2e 31 0e 76 65 72 73 69 6f 6e 3d   api=0.1.version=
00a0   31 2e 33 2e 31 30 c0 33 00 21 00 01 00 00 00 0a   1.3.10À3.!......
00b0   00 0d 00 00 00 00 13 88 04 68 6f 73 74 c0 1c c0   .........hostÀ.À
00c0   8c 00 1c 00 01 00 00 00 0a 00 10 2a 02 09 08 18   ...........*....
00d0   74 d3 41 80 22 da f0 af 08 f2 42 c0 8c 00 1c 00   tÓA."Úð¯.òBÀ....
00e0   01 00 00 00 0a 00 10 2a 02 09 08 18 74 d3 41 b4   .......*....tÓA´
00f0   c9 60 dc 1a 75 93 19 c0 8c 00 01 00 01 00 00 00   É`Ü.u..À........
0100   0a 00 04 c0 a8 00 0f                              ...À¨..

Multicast Domain Name System (response)
    Transaction ID: 0x2081
    Flags: 0x8400 Standard query response, No error
    Questions: 1
    Answer RRs: 6
    Authority RRs: 0
    Additional RRs: 0
    Queries
        _octoprint._tcp.local: type PTR, class ANY, "QM" question
    Answers
        _octoprint._tcp.local: type PTR, class IN, OctoPrint instance on host._octoprint._tcp.local
        OctoPrint instance on host._octoprint._tcp.local: type TXT, class IN
        OctoPrint instance on host._octoprint._tcp.local: type SRV, class IN, priority 0, weight 0, port 5000, target host.local
        host.local: type AAAA, class IN, addr 2a02:908:1874:d341:8022:daf0:af08:f242
        host.local: type AAAA, class IN, addr 2a02:908:1874:d341:b4c9:60dc:1a75:9319
        host.local: type A, class IN, addr 192.168.0.15
    [Unsolicited: True]
```

---

This is the my implementation, and it __is NOT recognized by Slic3r PE__:

[image]

```
me@host:~$ avahi-browse -ar
[... 189 lines cut ...]
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  // Setup MDNS responders
  MDNS.setHostProbeResultCallback(hostProbeResult);

  // Init the (currently empty) host domain string with 'esp8266'
  if ((!MDNSResponder::indexDomain(pcHostDomain, 0, "esp8266")) ||
      (!MDNS.begin(pcHostDomain))) {
    Serial.println(" Error setting up MDNS responder!");
    while (1) { // STOP
      delay(1000);
    }
  }
  Serial.println("MDNS responder started");

}

void loop(void) {

  // Allow MDNS processing
  MDNS.update();
}
```

vojtechkral, 2019-02-26:
@probonopd Thanks for the detailed data!

> Why is this not recognized in Slic3r PE? Is it because the A record is sent as an "additional record"? Or is it because we have no IPv6 AAAA record?

Neither of those should be a problem. But of course there might be a bug. I'll feed that response into my parser and have a look what's going on... I'll get back to you...

LaborEtArs, 2019-03-01:
The A record is placed in the additional section, because it wasn‘t asked for in the query itself. The missing AAAA shouldn’t be a problem, but if something is waiting for it might be anyways! As the currently online available responder version doesn’t send NSEC records, any querier, who is waiting for AAAA records might have a timeout here (eg. Chrome seems to have this).
However, the logs doesn’t fit exactly, they may be created by different setups: Once the service uses the hostname, while in the Wireshark dump it is independently named. But this should lead to any of the mentioned problems.
As Avahi sees the host, as well as the service, the question seems to be allowed: Might the problem be on the other side (the printer)?

vojtechkral, 2019-03-01:
@probonopd So I had a look and the parser parses the ESP's response correctly and extracts the IP correctly from the Addtional record. The problem seems to be the transaction ID. I don't remembder exatcly but I think followed the default DNS rules where the scanner randon-generates the ID and expect the response to match. Therefore, slic3r discards the response because of the non-matching transaction ID.

According to [this blog](http://imjac.in/ta/post/2016/08/24/writing-your-own-mdns.html) the ID should be zero for mDNS and indeed the zero-conf browser on Linux sends out requests with ID of zero.

I'll try to lookup an authoritative source on this and fix the implementation if indeed the ID should be zero.

LaborEtArs, 2019-03-01:
In fact, for mDNS  multicast responses the transaction ID 'MUST' be '0' (RFC 6762 18.1)

vojtechkral, 2019-03-01:
@probonopd I patched the Bonjour code, would you please re-test with the following build?
[Slic3rPE-1.42.0-alpha6+238-vk-octoprint-linux64-full-g8fd300c-201903011514.tar.gz](https://github.com/prusa3d/Slic3r/files/2920125/Slic3rPE-1.42.0-alpha6%2B238-vk-octoprint-linux64-full-g8fd300c-201903011514.tar.gz)

probonopd, 2019-03-01:
Sure enough it works :+1: Thank you very much @vojtechkral

vojtechkral, 2019-03-01:
Great, thanks for testing.

(Leaving issue open for now to track merging into `master`.)

bubnikv, 2019-03-01:
It will be part of alpha7. Closing.

bubnikv, 2019-03-02:
Reopening, as according to @vojtechkral the fix is not in the master yet.

commit d4ac57dd (2019-03-05): Bonjour: Fix: Set Query ID to zero, no ID checking in responses. Fix #1864
commit 96d09ce0 (2019-03-05): Bonjour: Fix: Set Query ID to zero, no ID checking in responses. Fix #1864
closed 2019-03-05 by vojtechkral