prusa3d/PrusaSlicer #1866: Feature Request: Gradient Mode & Random Mode like Palette 2

tbedore, 2019-02-25:
Just a feature request, on the palette 2 you have the ability to select the slice length, or have it randomly splice filaments into the extruder.  Would be nice if we could see that on the MMU2 upgrade.  Something similar to that.

flinke-flasche, 2020-03-14:
i would looking forward for such a feature.

flinke-flasche, 2020-03-23:
I’m looking forward for such a feature. 

Maybe it's possible to insert some values like

how often you like to change
which of these extruder you like to use -> 1, 2, 3, 4, 5 (checkboxes for each extruder)
min. Number of lines each color
max. Number of lines each color
purge on/off

bubnikv, 2020-03-23:
PrusaSlicer 2.2.0 already implements rainbow patterns with predefined color sequences. We will add the random rainbow patterns to PrusaSlicer 2.3.

flinke-flasche, 2020-03-23:
good to hear.
Can you show me how i can use the rainbow patterns?

bubnikv, 2020-03-23:
click on the cog wheel icon at the path preview layer slider.

flinke-flasche, 2020-03-23:
Perfect. thank you.

KnutssonDevelopment, 2020-04-29:
How about Color Gradient.

I would really like to see the possibility to do a gradient color change in the Z axis.

For it to work you would need to define starting color, finish color, start Z height, end Z height.

bubnikv, 2020-04-29:
Gradient? For which technology? Do you have a mixing extruder?

KnutssonDevelopment, 2020-09-03:
Nope. For the MMU2. Do a gradient color from one color to another and maybe with a transition color aswell.

The process I imagine the slicer doing is start with layers 100% in one color, and then mixing in layer in another color, until the second color is 100%.

The gradient would be in the z direction ofcourse.

I would expect to be able to set parameters as:
First Extruden/Fillament
Second Extruder/Fillament
Gradient change length in Z direction.

bubnikv, 2021-08-11:
We have implemented random ordering of the extruders with a fixed number of layers / fixed Z step in PrusaSlicer 2.4.0-alpha1.

We will not implement random number of layers per color now. We believe our implementation is sufficient.

The MMU2 technology requires the filament exchange to be performed over the wipe tower. @FidelCapo experimented with setting a minimum wipe over the wipe tower to achieve a rainbow effect. Unfortunately it did not produce nice results, as PrusaSlicer currently swaps filaments at the end of a layer while the first extrusion after the filament swap happens at the inner perimeter. To achieve at least some rainbow effect, we would have to start with the external perimeter after filament swap, which would require some additional code complications. 

Mosaic achieves quite some mixing of the filament along the splice as it is fed into a single extruder. MMU2 pulls the filament out first, thus there is much less of the 1st filament left in the nozzle to be mixed with the 2nd filament, thus the mixing effect will not be as pronounced as with Mosaic. We don't think it is worth the effort.

bubnikv, 2021-08-11:
[image]

flinke-flasche, 2021-09-14:
Thank you for implementing this cool feature.
Works perfectly for me.

Is it possible to add a Filament lenght to choose?
In my imagination you can print cool camouflage prints like in the Picture.
(i know that in reality the camouflage will be more different like in the picture :-D)

[image]

KnutssonDevelopment, 2021-09-14:
> I believe you can already experiment with the gradients by placing manual color changes and decreasing the amount of filament purge to a minimum, so that the wipe tower will be used for filament swap sequence, but not for filament purging.

I don´t think you are understanding the request. Doing hundreds of filament changes is not really doable by manual changes.

I am not sure how to make my point any more clear, but I will try.

Please make it so that you can set a starting filament Color/Filament, and a end Color/Filament, and a Z start height, and and end height.

Where the slicer gets to the start point it will initially start with a 100% of the start Color/Filament, then blend in first maybe every 5th layer of the destination Color/Filament, then every 4th layer, then every 3th layer, then every 2th layer, then every 2 ot of 3 layers, the 3 out of 5 layers, 4 out of 5 layer until 100% is the end Color/Filament.

This should of course be calculated from the Z start and end parameters.

I hope it makes sense. There is in this case no need to wipe, as the colors are going to be blended anyway.

closed 2026-09-29 by prusaslicer-bot[bot]