prusa3d/PrusaSlicer #1867: Future Request: estimated printing time variable for GCode

einencool, 2019-02-25:
Hello :-)

### Version
Version 1.42 Alpha 5

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 with MKS Gen L and Marlin 1.1.9

### Behavior
Hello, 
is it possible to get the Variable Slic3r PE uses to fill the estimated printing time at the end of the GCode?

[image]

I would like to take this variable and put a M117 Command with the estimated printing time to show it on the LCD.
Since my Printer doesn't support the M73 Command with the Counter, I would like to add these information automatically to my GCode.

For now I look into the GCode File, take the estimated printing time and put it with an 
M117 Printtime - xx h - xx m
into the Custom Start GCode from my file.
[image]

_Is this a new feature request?_
I think so :-)

It would be great if this would be possible, because the calculation of the printing time works like a charm, and there are only 1 or 2 minutes difference, also when the printing time is about several hours.
And I think there must be a chance to get this information easily because Slic3r already put this information into the GCode File...

Thank you in advance.
Greets Chris

dartrax, 2019-02-25:
Hello Chris,

welcome in the slic3r PE Community!

I'm missing that feature also, see #1739.
However, thanks for adding your vote on that ;-)

bubnikv, 2019-02-27:
Are you aware, that you can enter the following macros into the output file template?

print_time, normal_print_time, silent_print_time, used_filament,
extruded_volume, total_cost, total_weight, total_wipe_tower_cost, total_wipe_tower_filament

That may solve your request at least partially.

dartrax, 2019-02-27:
Hi,
yes I am, but thanks for the tip anyway. While that is nice to have, it does not help when you want the printer to show the estimated print time on its LCD. We need that macro inside the Gcode at the Layer Change Script so it can be combined with M117. And of course it should calculate the remaining print time, not the whole print time ;-)

einencool, 2019-02-27:
I'll give it a try today.
For me it's OK, because I can see in the bottom of the display the whole printing time, and in the status bar in the mittle the already printed time.

I tried different variables but all of them didn't work.

Do I have to set it in [ ] or do I need other signs?

einencool, 2019-02-27:
Well, I tried it, but the variables are not usable :-(
These are the failure messages, when I tried the variables...
Did I something wrong?

!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 2: Variable does not exist
M117 [print_time]
      ^
!!!!! End of an error report for the custom G-code template start_gcode

!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 2: Variable does not exist
M117 [normal_print_time]
      ^
!!!!! End of an error report for the custom G-code template start_gcode

bubnikv, 2019-02-27:
> Well, I tried it, but the variables are not usable :-(

> > Are you aware, that you can enter the following macros into the output
file template?

They are usable in the output file template, not custom G-code.

einencool, 2019-02-27:
Ah ok, i understand, yes that works, and for the first time it will be ok, but the other M117 command in the Start GCode would be way better to display.

But thank you, I'll give it a try :-)

rapkyt, 2025-04-11:
I also wanted a sense of progress other than the percentage the printer shows, and I tend to forget the height of the model I'm printing. So I added this custom G-code to show the estimated height of the model (it assumes that the print has a fixed layer height):

```
M117 estimated height: {total_layer_count * layer_height}
```

closed 2026-09-29 by prusaslicer-bot[bot]