prusa3d/PrusaSlicer #1869: [Slic3r 1.42.0 alpha4]: Lithophane - Layers missing

Flav1u, 2019-02-25:
As reported here [link](https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/lithophane-layers-missing-t27659.html#p128492), Slicr alpha seems to have trouble slicing lithophanes. 1.41.3 works fine.

dzieva, 2019-02-26:
I'm also having this problem with Slic3r version 1.42.0 alpha5

dzieva, 2019-02-28:
A sample of what occurs with lithophane
[image]

This problem does not happen in Simplify3D

bubnikv, 2019-03-01:
The only difference between the slicing algorithm in 1.41.3 and
1.42.0-alpha5 is the gap closing step, which was suppressed in 1.42.0. You
may simulate this gap closing step by applying a positive offset to each
slice:

[image: image.png]

From the code comments:

> Perform a safety offset to merge very close facets (TODO: find test case
for this)
> 0.0499 comes from https://github.com/slic3r/Slic3r/issues/959
> 0.0001 is set to satisfy GH #520, #1029, #1364

Please read the issues referenced to get the idea of why we are tinkering
with the gap closing parameter.

Please try to enter 0.05 into the "XY Size Compensation" field in alpha5
whether it will solve your issues. Your slices will be 50um thicker, so it
is not a solution, but it will help to diagnose your problem.

In the final 1.42.0 version, we will most likely make the gap closing
offset configurable, as it seems to be an issue to many.

Flav1u, 2019-03-01:
I tried different parameters with XY Size Compensation and nothing helped (it actually got worse, even larger layers missing). 

If the initial idea was to make meshes watertight with the morphological closing operation it probably was a very quick solution to a very complex problem. In fact I couldn’t find this sort of solution in literature at all. (E.g.: [A Practical Guide to Polygon Mesh Repairing](https://www.graphics.rwth-aachen.de/media/papers/eg2012_tutorial_meshrepair_021.pdf) or [Model Repair](http://www.pmp-book.org/download/slides/Model_Repair.pdf))

If the intention was to repair meshes (that are already watertight) this should require user interaction (on/off/set parameter) imho.

bubnikv, 2019-03-01:
Please provide an STL to test with, or better export the project (the 3mf file). Thank you.

> If the initial idea was to make meshes watertight with the morphological closing operation it probably was a very quick solution to a very complex problem.

Hmmm

> In fact I couldn’t find this sort of solution in literature at all. (E.g.: [A Practical Guide to Polygon Mesh Repairing](https://www.graphics.rwth-aachen.de/media/papers/eg2012_tutorial_meshrepair_021.pdf) or [Model Repair](http://www.pmp-book.org/download/slides/Model_Repair.pdf))

Repairing triangular models is a difficult task indeed. That's why there are not many software packages that do it right, for example the Netfabb.

We are not trying to compete with Netfabb. The task here is specific in that

1) We are working with slices. It is much easier to fix a slice than a 3D model.
2) We are extruding with a finite diameter extruder. That allows us to cut some corners in the slice accuracy. Namely, a morphological closing operation with a radius much smaller than the nozzle diameter will have insignificant effect to the print accuracy.
3) And lastly, it worked very well in many versions including Slic3r PE 1.41.3.

bubnikv, 2019-03-01:
All right, I am able to reproduce the issue with https://www.thingiverse.com/thing:2631187/files

bubnikv, 2019-03-01:
Running the lion model through the netfabb fixing service (integrated with Slic3r PE on Windows 10), the litophane is sliced correctly even with the current master. I will have to look into why the 1.41.3 handles this broken model correctly, while the 1.42.0-alpha5 does not.

dzieva, 2019-03-01:
Nothing happens using 0.05 in "XY Size Compensation"
[image]

Preview after...
[image]

Flav1u, 2019-03-01:
> Running the lion model through the netfabb fixing service (integrated with Slic3r PE on Windows 10), the litophane is sliced correctly even with the current master. I will have to look into why the 1.41.3 handles this broken model correctly, while the 1.42.0-alpha5 does not.

Intereseting. I imported and exported the lithophane stl in Meshmixer and now it worked in the alpha version too :)

dzieva, 2019-03-02:
I tested slic3r 1.42.0-alpha7 with the slice_closing_radius setting, but it does not fix the problem.
Any value assigned to slice_closing_radius that is non-zero only makes it worse.

slice_closing_radius = 0.049 (default)
[image]

slice_closing_radius = 0
[image]

slice_closing_radius = 0 with overhangs = true
[image]

bubnikv, 2019-03-11:
I hope I found the regression between 1.41.3 and 1.42.0-alpha7 in regard to closing gaps in the slices.
The issue should be fixed with c4829338453691371fc089d0aaf2cb9271417415

I would still appreciate, if you could attach a 3MF, so I can retest with your data.

bubnikv, 2019-03-17:
It will be fixed in beta. Closing.

closed 2019-03-17 by bubnikv

dzieva, 2019-03-17:
1.42.0-Beta fixes this problem.
Thank you.