prusa3d/PrusaSlicer #1870: Support Enforcers Don't Have Top Loops/Raft

screwyluie, 2019-02-26:
### Version
1.41.2+
1.42.0-alpha5

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
N/A
### Behavior
When you add support enforcers the supported area doesn't get the same roof as the rest of the supports. Don't know what to call it, raft? roof? the denser pattern on top of normal supports that contacts the surface of the model.

[image]
[image]

#### Project File (.3MF) where problem occurs
add .amf to the zip file and that's the exported project file from the alpha5... when loaded it doesn't seem to save the orientation so that's prolly not correct, you'll have to use the images as reference for the proper orientation.
[Breezy-ABL_Titan_with_EZABL_v3_RTP_UNIT.stl-3074179.zip](https://github.com/prusa3d/Slic3r/files/2903304/Breezy-ABL_Titan_with_EZABL_v3_RTP_UNIT.stl-3074179.zip)

bubnikv, 2019-03-01:
I am not sure I understand your problem. This is what I am getting from alpha7.

[image]

screwyluie, 2019-03-02:
Looks like you just set supports to "everywhere". This is an issue specific to support enforcers. Set it to build plate only and use the file I included

bubnikv, 2021-03-18:
Duplicate of #1997 #2801 #3831 #5748

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv