prusa3d/PrusaSlicer #1871: Infill pattern Archimedean causing total freeze at Infilling patterns..

DColl, 2019-02-26:
### Version
1.42 Alpha5 Win64
& 1.41.3 Win64

### on Windows 10 64bit
### Brand of printer not useful, occuring with different printer config

### Behavior
When slicing, the program freezes at "Infilling patterns". Totally bugging windows if not cancelled rapidly.

I added the "bugged_infill_pattern.ini" config file that creates the problem. Only by changing Top/bottom fill pattern to any choice except Archimedean solves the issue.

I believe the problem occurs with a combination of other settings, I've used Archimedean pattern previously without issue. Right now, I can't give better details.

#### Project File (.3MF) where problem occurs
[bugged_infill_pattern.zip](https://github.com/prusa3d/Slic3r/files/2903448/bugged_infill_pattern.zip)

[3DBenchy_test_base3mf.zip](https://github.com/prusa3d/Slic3r/files/2903453/3DBenchy_test_base3mf.zip)

DColl, 2019-02-26:
I think I've found the issue. It's occuring when I set the Top solid infill extrusion width to 0.1mm

Tried to set it at 1.5mm and behavior comes normal again.

[image]

bubnikv, 2019-03-07:
Reworked discretization step of Archimedean Chords infill to lower slicing time and memory requirements.

Fixed with 311df8ecfd93803bd51e6c2fa706a7e3d90bb338

It will be part of the 1.42.0-beta. Closing.

closed 2019-03-07 by bubnikv