prusa3d/PrusaSlicer #1874: [Slic3r 1.42.0 alpha5]: Crash on Cut Operation

simonkuehling, 2019-02-27:
### Version
1.42.0-alpha5+linux64

### Operating system type + version
Ubuntu 18.04.1 LTS

### Behavior
* _Describe the problem_
On some occasions (maybe 50%) Slic3r crashes completely on a "cut" operation to a model. This is model independent and happened to a large variety of different models.
* _Steps needed to reproduce the problem_
I have not found a way to reliably reproduce the issue - with the same model sometimes it works, next time it may fail...
* _Command Line Error Message_
`The program 'slic3r' received an X Window System error.`
`This probably reflects a bug in the program.`
`The error was 'BadMatch (invalid parameter attributes)'.`
`  (Details: serial 385801 error_code 8 request_code 155 minor_code 11)`
`  (Note to programmers: normally, X errors are reported asynchronously;`
`   that is, you will receive the error a while after causing it.`
`   To debug your program, run it with the --sync command line`
`   option to change this behavior. You can then get a meaningful`
`   backtrace from your debugger if you break on the gdk_x_error() function.)`

Is there a way to increase the debugging output to gather more information or a trace for you?

mkgin, 2019-02-27:
I noticed this also a couple weeks ago. I thing it is reproducable as follows, at least for me:

- Import stl wilth multiple objects
- "split into parts"
- sliced 
- check slice tab and go back, (not sure this is necessary)
- selecting and slicen object.
- crash

My error code is slightly different.

```
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 328716 error_code 8 request_code 154 minor_code 11)
```

I ran strace, there is about 100k lines without the ioctls. I attached the last 60000 lines where I believe the relevant stuff starts.

[strace_slic3r.gz](https://github.com/prusa3d/Slic3r/files/2910337/strace_slic3r.gz)

bubnikv, 2019-02-27:
A bugfix is already in the master, it is waiting for release. From the
upcoming change log:

> Crash on cut solved by switching the cut triangulation library to GLU
tesselator developed by SGI in 1994. While the application should not
crash, the cut may not get triangulated properly if the model is broken #105
<https://github.com/prusa3d/Slic3r/issues/105>#1349
<https://github.com/prusa3d/Slic3r/issues/1349> #1398
<https://github.com/prusa3d/Slic3r/issues/1398> #1442
<https://github.com/prusa3d/Slic3r/issues/1442> #1464
<https://github.com/prusa3d/Slic3r/issues/1464>.

simonkuehling, 2019-02-27:
@bubnikv awesome, thanks for the feedback! 

Closing this issue for now, then...

sgoll, 2019-03-09:
@bubnikv Does this commit also fix the following crash? These are the exact minimal steps required to reproducibly crash the "Cut" operation in 1.42.0-alpha7(+linux64):

1. Go to Preview (Ctrl+6)
2. Go back to 3D Editor Viewer (Ctrl+5)
3. Load STL (any file)
4. Select object and perform "Cut" operation (options do not matter)

So the crash is related to having displayed the Preview pane previously–even without any objects loaded.

lukasmatena, 2019-03-18:
@sgoll I believe there were several independent issues related to the cut function. The one related to switching between preview and 3d scene tabs seems NOT to be fixed in current master.

I looked into it a little bit and the bug appeared with https://github.com/prusa3d/Slic3r/commit/0a72a4e7ad7d6384be3ef8ca49edef9b483eee8d (which I verified by checking it out and commenting the three lines out). This might be something for @enricoturri1966 to look at.

enricoturri1966, 2019-03-18:
See also #1955

bubnikv, 2019-03-24:
It looks like the gtk / OpenGL does not like activating the OpenGL context if the window is not visible. The work around will be in the beta1. Closing.

closed 2019-03-24 by bubnikv