prusa3d/PrusaSlicer #1875: Custom/generic build plate image

wavexx, 2019-02-27:
I came to love the bed image which shown when I use the MK3. Combined with the pattern on the steel sheet, it makes precise placement of objects and application of gluestick a breeze.

I'd love to be able to set a generic image file to be shown, for any printer configured. I don't care about the entire bed model. Just showing the image overlay under the regular grid would be perfect already.

The reason is that I might have scratches/marks/obstacles on the build plate. It would be sooo much easier just to take a picture of the plate and set it so that I can manually avoid the damaged spots instead of doing this by estimation.

thisiskeithb, 2019-05-21:
Being able to load vector images of beds for other other printers would be a cool feature.

lukasmatena, 2019-07-31:
Implemented by @enricoturri1966. Closing.

closed 2019-07-31 by lukasmatena